enjoy-digital/litedram #6: Litedram is currently broken on the minispartan6+

mithro, 2017-08-13:
The minispartan6+ has sdram rather than DDR.

mithro, 2017-08-13:
TODO(@mithro): Put some type of info about the error messages / what is actually going wrong.

enjoy-digital, 2019-01-17:
This is solved by https://github.com/enjoy-digital/litedram/commit/b4c552a77f8aef2f0e153075c4b51acf40977eee. (Tested on https://github.com/enjoy-digital/chubby75).

closed 2019-01-17 by enjoy-digital

mithro, 2019-01-17:
@enjoy-digital Did you actually test on the minispartan6+ or just on the chubby75?

enjoy-digital, 2019-01-18:
I reproduce on chubby75 (firmware loaded in dram + BIST hanging) and verified it was working correctly with https://github.com/enjoy-digital/litedram/commit/b4c552a77f8aef2f0e153075c4b51acf40977eee)

mithro, 2019-01-18:
Will find a minispartan6+ to test on...

enjoy-digital, 2019-01-18:
I just tested on the minispartan6+ (the files are also attached if you want to test on a minispartan6+ xc6slx25):
````
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: LM32 /  80MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Jan 18 2019 14:59:21

BIOS CRC passed (a21ce08f)
Initializing SDRAM...
SDRAM now under hardware control
Memtest OK
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
[TERM] Received firmware download request from the device.
[TERM] Uploading firmware.bin (11564 bytes)...
[TERM] Upload complete (7.6KB/s).
[TERM] Booting the device.
[TERM] Done.
Executing booted program at 0x40000000

Minispartan6 CPU testing software built Jan 18 2019 15:01:44
RUNTIME>WR_SPEED(Mbps) RD_SPEED(Mbps)  TESTED(MB)       ERRORS
          1231           1152           0            0
          1231           1152           3            0
          1231           1152           6            0
          1231           1152           9            0
          1231           1152          12            0
          1231           1152          15            0
          1231           1152          18            0
          1231           1152          21            0
          1231           1152          24            0
          1231           1152          27            0
WR_SPEED(Mbps) RD_SPEED(Mbps)  TESTED(MB)       ERRORS
          1231           1152          30            0
          1231           1152          33            0
          1231           1152          36            0

RUNTIME>sdram_bist 256 1
WR_SPEED(Mbps) RD_SPEED(Mbps)  TESTED(MB)       ERRORS
          1208           1175           0            0
          1202           1181           7            0
          1202           1181          13            0
          1202           1181          19            0
          1202           1181          25            0
          1202           1181          31            0
          1202           1181          37            0
          1202           1181          43            0
          1202           1181          49            0
          1202           1181          55            0
WR_SPEED(Mbps) RD_SPEED(Mbps)  TESTED(MB)       ERRORS
          1202           1181          61            0
          1202           1181          67            0
          1202           1181          73            0
RUNTIME>
RUNTIME>
````

IIRC, before executing a firmware from SDRAM was almost immediately crashing.  I tested running  the firmware from SDRAM + SDRAM BIST for more than one hour.

[minispartan6_sdram_test.zip](https://github.com/enjoy-digital/litedram/files/2773582/minispartan6_sdram_test.zip)

mithro, 2019-01-18:
Oh. Awesome!