enjoy-digital/litedram #8: Add some information about performance / bandwidth

mithro, 2017-08-13:
It would be good if the docs had some information about performance of litedram in different configurations.

Maybe it could be added [to my spreadsheet here](https://docs.google.com/spreadsheets/d/1q5CUI289TjGw9c66wAesOHlvH8RVo-OK6_yJ4btzZPA/edit#gid=150757850).

mithro, 2017-08-13:
We know that in the [HDMI2USB-litex-firmware](https://github.com/timvideos/HDMI2USB-litex-firmware)  on the Numato Opsis we currently work fine with;
 * 2 x 720p50 YUV outputs
 * 1 x 720p50 YUV  input
 * 1 x encoder?? (720p25 I think?)

But if we go to the following, we start getting underflows on the outputs;
 * 2 x 720p60 YUV outputs
 * 1 x 720p60 YUV input
 * 1 x encoder?? (720p30 I think?)

Looking at my ["graphics resolution spreadsheet"](https://docs.google.com/spreadsheets/d/1JvBYgGgWN35BF30zXw76Jp7c4_SLcykoOMWNcuDTUv4/edit#gid=720646227) we know with 720 @ YUV at 16bits;
 * p50 is 737 Mbits/second
 * p60 is 885 Mbits/second

Thus, the working 720p50 configure is somewhere around ~2211 Mbit/second and the 720p60 configure which is getting underflows is around ~2655 Mbit/second.

[The Opsis is configured with 16bit width 1:4 DDR running at 400MHz](https://github.com/timvideos/HDMI2USB-litex-firmware/blob/master/targets/opsis/base.py#L96-L131).

enjoy-digital, 2020-01-28:
This will be covered by the benchmark tests.

closed 2020-01-28 by enjoy-digital