enjoy-digital/litedram #14: ddr calibration fails when SoC temperature is hot

xobs, 2018-07-19:
When die temp is high: 70.0107 (0x0ae5)

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv / 100MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Jul 19 2018 15:06:12

BIOS CRC passed (5bf71210)
Initializing SDRAM...
Read bitslip: 3
Read delays scan:
m0: 00000000000000000000000000000000
m1: 00000000001111111111111000000000
m2: 00000000001111111111111000000000
m3: 00000000001111111111111100000000
Read delays: 3:10-24  2:10-24  1:10-24  0:32-33  completed
Memtest bus failed: 156/256 errors
Memtest data failed: 524288/524288 errors
Memtest addr failed: 8192/8192 errors
Memory initialization failed
```

With the latest release, the board fails to boot:

```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
 SoC BIOS / CPU: VexRiscv / 100MHz
(c) Copyright 2012-2018 Enjoy-Digital
(c) Copyright 2007-2018 M-Labs Limited
Built Jul 19 2018 15:34:22

BIOS CRC passed (317e43f7)
Initializing SDRAM...
Read bitslip: 2
Read delays scan:
m0: 00000000000000000000000000000000
m1: 11111111000000000000000000000000
m2: 11111111000000000000000000000000
m3: 111111110000
```

enjoy-digital, 2018-07-19:
Thanks, i'll do the test on the 35T version. That would be interesting for me to also have the logs when it's working with 0x0ae5.

xobs, 2018-08-03:
This was likely due to a hardware fault caused by hand-reworking the board with a T100 part.

closed 2018-08-03 by xobs