enjoy-digital/litedram #22: Out of order interface is incomplete

JohnSully, 2018-08-10:
This is more of an FYI that out of order is not yet fully completed.  Specifically the write interface must also output the bank its using.  In addition OO is disabled in the frontend crossbar.

The interface is also not stable.  If we implement re-ordering within banks (e.g. write batching) then we'll have to move to a tag system to track operations.

enjoy-digital, 2018-08-10:
Yes thanks. I just wanted to integrate your changes. We'll try to add complete reordering later.

JohnSully, 2018-08-10:
I've integrated with your changes and pushed the remaining work to the outoforder branch.  This is the full code needed to use reordering.  I'll send another pull request once I'm happy the integration work didn't cause any bugs.

enjoy-digital, 2018-08-11:
Your pull request is merged.

closed 2018-08-11 by enjoy-digital