enjoy-digital/litedram #35: BL8 1:2 hangs with UART bridge

JohnSully, 2018-09-06:
What I know so far:

Only repros in 1:2 mode works fine in 1:4
Is sensitive to the frequency (design meets timing)
Writes succeed, only reads fail.

JohnSully, 2018-09-06:
Seems to be related to the time since reset.  The closer you are to a reset the longer it works for.   It also takes an entire second for initialization to complete which seems excessive.  I wonder if a timer is over/underflowing.

JohnSully, 2018-09-06:
This is caused by a conflict with the on-board JTAG of the Arty S7.  It uses the same FTDI chip as the USB uart and I suppose its conflicting with timing.  The issue goes away when you close debug server in vivado.

enjoy-digital, 2018-09-06:
Strange, but still interesting to know in case i have the same issue.

JohnSully, 2018-09-06:
So this still happens at higher clock frequencies even with an external JTAG adapter.
Worked at 125Mhz but not 150mhz core clock (1:2 controller).

Reading back the identifier works fine with no corruption so I don't think its a noise issue.

JohnSully, 2018-09-06:
Debug Output from calibration:

write 00000000 @ e0004000
Release reset
write 00000000 @ e000400c
write 00000000 @ e0004010
write 0000000c @ e0004000
Bring CKE high
write 00000000 @ e000400c
write 00000000 @ e0004010
write 0000000e @ e0004000
Load Mode Register 2, CWL=5
write 00000200 @ e000400c
write 00000002 @ e0004010
write 0000000f @ e0004004
write 00000001 @ e0004008
Load Mode Register 3
write 00000000 @ e000400c
write 00000003 @ e0004010
write 0000000f @ e0004004
write 00000001 @ e0004008
Load Mode Register 1
write 00000006 @ e000400c
write 00000001 @ e0004010
write 0000000f @ e0004004
write 00000001 @ e0004008
Load Mode Register 0, CL=6, BL=8
write 00000520 @ e000400c
write 00000000 @ e0004010
write 0000000f @ e0004004
write 00000001 @ e0004008
ZQ Calibration
write 00000400 @ e000400c
write 00000000 @ e0004010
write 00000003 @ e0004004
write 00000001 @ e0004008
write 00000001 @ e0004000
bitslip 0: |write 00000001 @ e0008004
write 00000001 @ e0008008
write 00000001 @ e0008010
write 00000002 @ e0008004
write 00000001 @ e0008008
write 00000001 @ e0008010
write 3c6ef35f @ 40000000
write 3c88596c @ 40000004
write 3ca1bf79 @ 40000008
write 3cbb2586 @ 4000000c
write 3cd48b93 @ 40000010
write 3cedf1a0 @ 40000014
write 3d0757ad @ 40000018
write 3d20bdba @ 4000001c
write 3d3a23c7 @ 40000020
write 3d5389d4 @ 40000024
write 3d6cefe1 @ 40000028
write 3d8655ee @ 4000002c
write 3d9fbbfb @ 40000030
write 3db92208 @ 40000034
write 3dd28815 @ 40000038
write 3debee22 @ 4000003c
Traceback (most recent call last):
  File "test_sdram_init.py", line 77, in <module>
    errors = check_pattern(nwords)
  File "test_sdram_init.py", line 47, in check_pattern
    if wb.read(wb.mems.main_ram.base + 4*i) != seed_to_data(i):
  File "/usr/local/lib/python3.5/dist-packages/litex-0.2.dev0-py3.5.egg/litex/soc/tools/remote/litex_client.py", line 45, in read
    packet = EtherbonePacket(self.receive_packet(self.socket))
  File "/usr/local/lib/python3.5/dist-packages/litex-0.2.dev0-py3.5.egg/litex/soc/tools/remote/etherbone.py", line 361, in receive_packet
    chunk = socket.recv(header_length - len(packet))
socket.timeout: timed out

JohnSully, 2018-09-06:
Sleeping 0.2 seconds after writing the test pattern appears to stop it from hanging (fails with 0.1).  Calibration doesn't find any valid configs but that may be unrelated. 

Interesting issue, I'm not sure what's causing it.

enjoy-digital, 2018-09-06:
OK, i'll try to see if i can reproduce it.

JohnSully, 2018-09-06:
Repros on my Arty S7-50 board.  As well as custom hardware with an xc7a50t-2csg324I part.

JohnSully, 2018-09-08:
Seems to be fixed with l2 cache is reenabled.  Still fails to calibrate though.