enjoy-digital/litedram #39: Multirank: add dynamic ODT

enjoy-digital, 2018-09-10:
To support higher frequencies in dual/quad rank mode, we will need to drive ODT dynamically.
Useful information can be found in:
- TN-41-08: Design Guide for Two DDR3-1066 UDIMM Systems
- TN-04-54: High-Speed DRAM Controller Design

softerhardware, 2018-09-10:
Hi Florent,

I see the following error now after updating everything to try multirank. I see this error now even when trying to build an existing target such as genesys2_config.py. 

I'm trying to test on a customer design that has a multirank DDR3. Last week I was able to build a test where I ignored the other ranks, but it did not work. I need to add more debug today. The problem may just be an error in my use of the IP. I've attached the config file and update modules.py to target this customer DDR3. Please let me know if you think they are okay.

I will also try this new multirank support once I can get past the error below. How should the ODT signals be connected for now?

Best Regards,

Steve
[ranksupport.tar.gz](https://github.com/enjoy-digital/litedram/files/2367948/ranksupport.tar.gz)

 

python3 s7_coregen.py targets/nl991_config.py

Traceback (most recent call last):
  File "s7_coregen.py", line 504, in <module>
    main()
  File "s7_coregen.py", line 467, in main
    soc = LiteDRAMCore(platform, core_config, integrated_rom_size=0x6000)
  File "s7_coregen.py", line 322, in __init__
    controller_settings=controller_settings)
  File "/home/shaynal/enjoy-digital/litex/litex/soc/integration/soc_sdram.py", line 56, in register_sdram
    phy, geom_settings, timing_settings, **kwargs)
  File "/home/shaynal/enjoy-digital/litex/litex/soc/integration/soc_sdram.py", line 19, in __init__
    phy.settings.dfi_databits, phy.settings.nphases)
  File "/home/shaynal/enjoy-digital/litedram/litedram/dfii.py", line 42, in __init__
    inti = dfi.Interface(addressbits, bankbits, nranks, databits, nphases)
  File "/home/shaynal/enjoy-digital/litedram/litedram/phy/dfi.py", line 45, in __init__
    Record.__init__(self, layout)
  File "/home/shaynal/enjoy-digital/migen/migen/genlib/record.py", line 106, in __init__
    finst = Record(fsublayout, prefix + fname, **kwargs)
  File "/home/shaynal/enjoy-digital/migen/migen/genlib/record.py", line 103, in __init__
    finst = Signal(fsize, name=prefix + fname, **kwargs)
  File "/home/shaynal/enjoy-digital/migen/migen/fhdl/structure.py", line 384, in __init__
    reset = Constant(reset, (self.nbits, self.signed))
  File "/home/shaynal/enjoy-digital/migen/migen/fhdl/structure.py", line 295, in __init__
    raise TypeError("Width must be a strictly positive integer")
TypeError: Width must be a strictly positive integer

enjoy-digital, 2026-11-10:
Status: ODT is still statically asserted on all the ranks by the controller (`multiplexer.py`). While looking at this, it also matters in single rank: on DDR3 with RTT_NOM/RTT_WR enabled (LiteDRAM's default, 60 ohm), the DRAM terminates during its own READs, which JEDEC does not allow ("RTT must be disabled at least one-half clock cycle before the READ preamble by driving the ODT ball LOW", Micron DDR3L datasheet, "ODT Off During READs"). DDR4 turns its termination off during READs by itself.

#485 adds `ControllerSettings(with_dynamic_odt=True)` (DDR3, single rank, opt-in for now): ODT asserted with each WRITE and held for ODTH8, deasserted otherwise. It is the base for the multi-rank target/non-target ODT (TN-41-08) of this issue, which still needs a multi-rank board to be validated: keeping this open for that part.