enjoy-digital/litedram #47: Data corruption when out of order ports accessed simultaneously.

JohnSully, 2018-09-27:
When I have an in-order port and an out-of-order port, traffic on the out of order port corrupts data on the in order port.  Issue goes away when traffic is stopped on the out of order port.

enjoy-digital, 2018-09-28:
I'm not sure we want to support mixing both. I just change the API to prevent that. (https://github.com/enjoy-digital/litedram/commit/5984eaa6da4f8d8a631d22f28faed0a500415096)

JohnSully, 2018-09-28:
Also fundamentally I don't see why it shuoldn't be allowed to have both.  If a bank is locked the out of order port simply can't use it until unlocked.

enjoy-digital, 2018-09-28:
Current out-of-order is half implemented, it just exposes the bank to user. We first need to implement https://github.com/enjoy-digital/litedram/issues/30 and then ask ourselve out to support both simultaneously.

closed 2018-09-28 by enjoy-digital