enjoy-digital/litedram #51: Add tCCD for older DDR technologies

mithro, 2018-10-02:
It seems that litedram requires a tCCD value on modules now. From @JohnSully and https://github.com/timvideos/litex-buildenv/issues/77#issuecomment-426125038

 * For DDR and SDR tCCD is 1 cycle.
 * LPDDR depends on the module. It’s burst length/2.

mithro, 2018-10-02:
Seems to be currently broken broken on DDR and SDRAM,
```
- make firmware check (mimasv2 base lm32 firmware)
---------------------------------------------
mkdir -p build/mimasv2_base_lm32/
time python -u ./make.py --platform=mimasv2 --target=base --cpu-type=lm32 --iprange=192.168.100     --no-compile-gateware \
		2>&1 | tee -a build/mimasv2_base_lm32//output.20181002-050444.log; (exit ${PIPESTATUS[0]})
Traceback (most recent call last):
  File "./make.py", line 164, in <module>
    main()
  File "./make.py", line 123, in main
    soc = get_soc(args, platform)
  File "./make.py", line 57, in get_soc
    soc = SoC(platform, ident=SoC.__name__, **soc_sdram_argdict(args), **dict(args.target_option))
  File "/home/travis/build/mithro/litex-buildenv/targets/mimasv2/base.py", line 241, in __init__
    controller_settings=controller_settings)
  File "/home/travis/build/mithro/litex-buildenv/third_party/litex/litex/soc/integration/soc_sdram.py", line 60, in register_sdram
    phy, geom_settings, timing_settings, **kwargs)
  File "/home/travis/build/mithro/litex-buildenv/third_party/litex/litex/soc/integration/soc_sdram.py", line 27, in __init__
    phy.settings, geom_settings, timing_settings, **kwargs)
  File "/home/travis/build/mithro/litex-buildenv/third_party/litedram/litedram/core/controller.py", line 66, in __init__
    settings)
  File "/home/travis/build/mithro/litex-buildenv/third_party/litedram/litedram/core/bankmachine.py", line 87, in __init__
    precharge_time = write_latency + settings.timing.tWR - 1 + settings.timing.tCCD # AL=0
TypeError: unsupported operand type(s) for +: 'int' and 'NoneType'
```

enjoy-digital, 2018-10-02:
Done here: https://github.com/enjoy-digital/litedram/commit/5b02791580db9d1bae64b6524b7aca3540d89937. Tested with test/test_targets in LiteX.

closed 2018-10-02 by enjoy-digital