enjoy-digital/litedram #52: Add a test design for each of the DDR technologies

mithro, 2018-10-02:
It would be good to have at least one test for each of the technology group;
 * SDRAM
 * DDR
 * LPDDR
 * DDR2
 * DDR3

I believe DDR3 is the primary one covered by tests at the moment?

enjoy-digital, 2018-10-02:
I'll try to add that to the examples and require that examples are building correctly before integrating changes.
This should prevent issues like https://github.com/enjoy-digital/litedram/issues/51.

enjoy-digital, 2018-10-02:
https://github.com/enjoy-digital/litex/blob/master/test/test_targets.py should cover most of this. DDR and LPDDR are missing, i'll try to add more configurations to test.

enjoy-digital, 2020-02-20:
This is now covered by the new Travis-CI tests that are testing all SDRAM modules.

closed 2020-02-20 by enjoy-digital