enjoy-digital/litedram #58: List various usecases, find best architecture and document how to get the best efficiency.

enjoy-digital, 2018-10-11:
The controller can be used on various systems which all have specific needs. The current architecture try to be simple and give reasonable efficiency/performance for most of the usecases.

Until now, performance was good on access patterns that were used. (mostly for linear data buffers and store CPU code). We still need to do https://github.com/enjoy-digital/litedram/issues/8 and https://github.com/enjoy-digital/litedram/issues/57 to have better metrics.

With more users, some new usecases appear with different access patterns and we need to understand what are the eventual bottlenecks and the way to improve this.

enjoy-digital, 2018-10-11:
@JohnSully: can you describe what kind of access pattern your system is doing? (without going into implementation details for now).  From what i understand, you are reading a random location and just after rewrite that location with an updated data. Is the location always random? do you need the result of the previous read to know where will happen the next read? Same for writes?

I just like to understand if some optimizations can be first done on the user side before touching too much the internals of the controller. If optimizations can be done on the user side, we could document them so that new users are aware of the limitations and prevent falling in a case where efficiency is low.

JohnSully, 2018-10-11:
I run a multi-threaded application that does a read, computes, and writes to the same location.  The next location is based upon the results of the prior read.  However the latency here is hidden by having multiple threads so the memory controller always has work to do.

sergachev, 2019-07-18:
I use litedram on Artix 7 with DDR3, 16b interface, 100MHz system clock, 4:1. Memory is written simultaneously by 8 independent ~600 Mbps sources, each has own destination buffer and own LiteDRAMDMAWriter, data width matched to dram native port width. If sources write data as soon as it is available (DRAM access becomes quite random), up to 4 of them can work correctly, meaning 2.4 Gbps. If every source gets an additional FIFO and only writes in bursts (after its FIFO collects some data), all 8 work correctly, meaning 4.8 Gbps.

Needless to say how satisfied I am by litedram (well, litex in general, but litedram especially)!

enjoy-digital, 2019-07-18:
Thanks @sergachev for your feedback! Is your project open-source? (just curious to know more if it's the case)

sergachev, 2019-07-18:
>  Is your project open-source? (just curious to know more if it's the case)

Currently no, but I will discuss with the client which parts can be open.
I will definitely submit pull requests for all general purpose features I add to litex.
For example, this bandwidth test I can submit just as a configurable benchmark.

enjoy-digital, 2019-07-18:
Thanks already for the feedback, additional tests/benchmarks are welcome :)

enjoy-digital, 2020-01-28:
This will be covered by the benchmark tests.

closed 2020-01-28 by enjoy-digital