enjoy-digital/litedram #80: Simulation is broken

dh73, 2019-05-10:
Hi,
I'm trying to test a custom configuration in simulation. Before that, I went to execute sim.py without any change and found couple of issues:
1) The mem_1.init is not referenced (copied) anywhere on sim.py
```bash
WARNING: File mem_1.init referenced on $PATH/litex/litedram/examples/sim/litedram_core.v at line 15687 cannot be opened for reading. Please ensure that this file is available in the current working directory.
```
2) There's a X prop when litedram_core releases the user reset and introduces an unknown state to the DDR3 model. 
```bash
top_tb.dut.ddr3.main: at time 530100.0 ps ERROR: CK and CK_N are not allowed to go to an unknown state.
```
[image]

3) I'm not sure if I'm doing something wrong or there's an extra step needed, but as far as I goes, I haven't had any successful simulation result. As you can see, init_done is never asserted, so the main fsm got stuck on idle states forever. Tracing back, that signal ends asserted by a CSR, but a) There's no CPU or register write interface and b) Both init memories are empty by default.

4) There are a lot of redundant writes to variables, therefore the generated code gets verbose. In this excerpt from *top.v* lines 335 to 341, the variable *fsm_next_state* is written twice, along with the declaration that is initialized. This can be improved.
```verilog
always @(*) begin
	sdram_checker_addr_gen_ce <= 1'd0;
	sdram_checker_sink_sink_valid <= 1'd0;
	fsm0_next_state <= 2'd0; // Already initialized by 0
	sdram_checker_cmd_counter_fsm0_next_value <= 24'd0;
	sdram_checker_cmd_counter_fsm0_next_value_ce <= 1'd0;
	fsm0_next_state <= fsm0_state; // Written again
```

Thank you.

enjoy-digital, 2019-05-10:
Hi @dh73,

sorry, the simulation was here just as an example on how to simulate with xsim for someone and is indeed not working as it it integrated. To avoid too long simulations we wre using a specific module to do standalone initialization of the DDR3 but it is not integrated here.

If you want to do sytem-level simulation, the best is to generate the same core you will run on hardware and integrate the DDR3 model as it is done here on the actual simulation. It will be a bit slow since the the CPU will be integrated and will do the initialization, but will probably be easier than using this simulation.

To respond to the others questions:
2) This is probably happening when the reset is released, i'm not sure we can avoid that.
4) This related to the way Migen generates code: the last affectation wins. This is not an issue, and we use this property in Migen, but yes would have been done better in manual code.

Before your message, i was going to remove this simulation since this is confusing so i will remove it. As recommended, if you want to system-level simulation, you need to generate the same core you will use for your board and simulate it with you traditionnal flow.

Regards,

Florent

dh73, 2019-05-15:
Hi Florent,
Thank you for the explanation, it makes perfect sense now.

closed 2019-05-15 by enjoy-digital