enjoy-digital/litedram #88: pin sdram_dm[1] stuck at GND can this be ignored?

AnttiLukats, 2019-08-04:
Warning (13410): Pin "sdram_dm[1]" is stuck at GND File: X:/GIT/litex-boards/litex_boards/partner/targets/soc_basesoc_c10lprefkit/gateware/top.v Line: 3435

can this warning be ignored?

memtest reports OK, but having 1 out of 2 DM pins stuck GND does not sound like a warning to be ignored ?

AnttiLukats, 2019-08-04:
my guess it was because wrong platform file that only included constraints for 15 pins, one was missing, so my guess is that such warning should not be ignored.. fixing my platform file

enjoy-digital, 2019-08-04:
Ah ok, i understand. I tried to reproduce on de0nano but was not able to see the warning. We could try to catch these issues directly in litex doing some assertions on length of dm vs length of dq.

enjoy-digital, 2019-08-04:
This will now be detected with https://github.com/enjoy-digital/litedram/commit/8573c22cc1a10c89dcff3e838387a676c3b2ac5e.

closed 2019-08-04 by enjoy-digital