enjoy-digital/litedram #96: Problems instantiating on ice40 board

niklasnisbeth, 2019-12-30:
I have made an SDRAM plug-in board for the Lattice ICE40 HX8K EVB. It uses the AS4C16M16 which is already supported.

I am trying to make a simple Litex project to use it, but I get a synthesis error:

```
$ python3 ice40hx8k_litedram_nn.py 
lxbuildenv: v2019.8.19.1 (run ice40hx8k_litedram_nn.py --lx-help for help)
<__main__.Platform object at 0x7fb78610b7f0>
{'cpu_type': None, 'cpu_variant': None, 'integrated_rom_size': 0, 'integrated_sram_size': 0}
ERROR: Conflicting init values for signal 1'1 (\soc_sdram_master_p0_act_n = 1'1, \soc_sdram_choose_req_want_activates = 1'0).
Traceback (most recent call last):
  File "ice40hx8k_litedram_nn.py", line 224, in <module>
    main()
  File "ice40hx8k_litedram_nn.py", line 220, in main
    vns = builder.build()
  File "/home/niklas/hack/litex/notbuildenv/deps/litex/litex/soc/integration/builder.py", line 185, in build
    toolchain_path=toolchain_path, **kwargs)
  File "/home/niklas/hack/litex/notbuildenv/deps/litex/litex/soc/integration/soc_core.py", line 452, in build
    return self.platform.build(self, *args, **kwargs)
  File "/home/niklas/hack/litex/notbuildenv/deps/litex/litex/build/lattice/platform.py", line 34, in build
    return self.toolchain.build(self, *args, **kwargs)
  File "/home/niklas/hack/litex/notbuildenv/deps/litex/litex/build/lattice/icestorm.py", line 189, in build
    _run_script(script)
  File "/home/niklas/hack/litex/notbuildenv/deps/litex/litex/build/lattice/icestorm.py", line 121, in _run_script
    raise OSError("Subprocess failed")
OSError: Subprocess failed
```

I have created a [repo here ](https://github.com/niklasnisbeth/ice40-litedram/)with a small example file and the build output checked in. I'm using yosys (Yosys 0.9+932 (git sha1 9e6632c4, clang 9.0.0 -fPIC -Os).

enjoy-digital, 2019-12-30:
Thanks for the report and repo. You can safely use your patch for now since act_n is not used on SDRAM. The issue is in fact probably the same we encountered on ECP5 (https://github.com/YosysHQ/yosys/pull/850) and that should probably also be applied in Yosys on iCE40 too. @daveshah1 what are your thoughts, should it also be lowered to a Warning as done on ECP5, should we create a Yosys issue for that?

niklasnisbeth, 2019-12-31:
Thanks for the reply. Do I understand correctly that this looks entirely like a Yosys issue that should be fixed on that side?

niklasnisbeth, 2019-12-31:
I was able to copy the commit you linked to in the ice40 techlib and can confirm that this fixes the issue. I have made a PR with Yosys.

enjoy-digital, 2020-01-23:
Yosys PR has been merged, we can close this.

closed 2020-01-23 by enjoy-digital