enjoy-digital/litedram #99: Occasional "Memtest bus failed"

sergachev, 2020-01-16:
I have a design targeting Artix-7 (several sizes, speed grades -1, -2) and MT41K128M16 (256 MB DDR3). It works very well in general, but from time to time implementations fail with the error below. A re-implementation of the same design most often works perfectly. The failure is perfectly stable: it always gives the same amount of errors (144) and never passes the test. The successful implementations are also perfectly stable (never fail). Every implementation - also failing ones - has formally valid timings ("All user specified timing constraints are met" in vivado log).

```
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |01111111111110000000000000000000| delays: 07+-06
m0, b1: |00000000000000001111111111110000| delays: 22+-06
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: 07+-06
m1, b0: |11111111110000000000000000000000| delays: 05+-05
m1, b1: |00000000000000001111111111000000| delays: 21+-05
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: 05+-05
SDRAM now under hardware control
Memtest bus failed: 144/256 errors

```

sergachev, 2020-01-16:
Delay numbers vary from boot to boot, and a successful implementation may output exactly same numbers as a failing one:

```
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |01111111111110000000000000000000| delays: 07+-06
m0, b1: |00000000000000001111111111110000| delays: 22+-06
m0, b2: |00000000000000000000000000000000| delays: -
m0, b3: |00000000000000000000000000000000| delays: -
m0, b4: |00000000000000000000000000000000| delays: -
m0, b5: |00000000000000000000000000000000| delays: -
m0, b6: |00000000000000000000000000000000| delays: -
m0, b7: |00000000000000000000000000000000| delays: -
best: m0, b0 delays: 07+-06
m1, b0: |11111111110000000000000000000000| delays: 05+-05
m1, b1: |00000000000000001111111111000000| delays: 21+-05
m1, b2: |00000000000000000000000000000000| delays: -
m1, b3: |00000000000000000000000000000000| delays: -
m1, b4: |00000000000000000000000000000000| delays: -
m1, b5: |00000000000000000000000000000000| delays: -
m1, b6: |00000000000000000000000000000000| delays: -
m1, b7: |00000000000000000000000000000000| delays: -
best: m1, b0 delays: 05+-05
SDRAM now under hardware control
Memtest OK

```

enjoy-digital, 2020-01-16:
@sergachev: i think i already saw that and that was caused by the clocking or reset. Can you provide the CRG you are using?

sergachev, 2020-01-16:
```
class ClockResetGen(Module, AutoCSR):
    def __init__(self, platform, eth_pads):
        cd = self.clock_domains
        cd.cd_125 = ClockDomain()
        cd.cd_125_90 = ClockDomain(reset_less=True)
        cd.cd_clk50 = ClockDomain()
        cd.cd_clk10 = ClockDomain()

        self.cd_125.clk.attr.add("keep")
        self.cd_125_90.clk.attr.add("keep")
        self.cd_clk50.clk.attr.add("keep")

        self.submodules.pll1 = pll1 = S7MMCM(speedgrade=speed_grade)
        pll1.register_clkin(platform.request("clk50"), 50e6)
        pll1.create_clkout(self.cd_125, 125e6)
        pll1.create_clkout(self.cd_125_90, 125e6, phase=90)
        pll1.create_clkout(self.cd_clk50, 50e6)
        pll1.create_clkout(self.cd_clk10, timestamp_clock_freq)

        cd.cd_sys = ClockDomain()
        cd.cd_sys4x = ClockDomain(reset_less=True)
        cd.cd_sys4x_dqs = ClockDomain(reset_less=True)
        cd.cd_clk200_pll2 = ClockDomain()

        self.cd_sys.clk.attr.add("keep")
        self.cd_sys4x.clk.attr.add("keep")
        self.cd_sys4x_dqs.clk.attr.add("keep")

        self.submodules.pll2 = pll2 = S7MMCM(speedgrade=speed_grade)
        self.comb += pll2.reset.eq(0)
        pll2.register_clkin(self.cd_clk50.clk, 50e6)
        pll2.create_clkout(self.cd_sys, sys_clk_freq)
        pll2.create_clkout(self.cd_sys4x, 4 * sys_clk_freq)
        pll2.create_clkout(self.cd_sys4x_dqs, 4 * sys_clk_freq, phase=90)
        pll2.create_clkout(self.cd_clk200_pll2, 200e6)

        self.submodules.idelayctrl = S7IDELAYCTRL(self.cd_clk200_pll2)

```

sergachev, 2020-11-12:
The problem was indeed caused by improper PLL resets.

closed 2020-11-12 by sergachev

enjoy-digital, 2020-11-12:
@sergachev: great you solved it, thanks for the feedback.