enjoy-digital/litedram #100: Support 1:4 frequency ratio on ECP5

rowanG077, 2020-01-22:
I'm not sure if this is technically possible but it would be really nice if this component could support 1:4 frequency ratio for the ECP5.

enjoy-digital, 2020-01-27:
It's not directly possible with the ODDRX2*/IDDRX2* but we could use the same technique we used for Spartan6: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s6ddrphy.py#L407. This was a bit hacky it but could be a way while doing this to clean it up :)

enjoy-digital, 2021-01-08:
This would be a nice contribution but this hasn't been planned to work on this.

closed 2021-01-08 by enjoy-digital

thirtythreeforty, 2021-06-30:
Hi, could you expand on this a little? I am interested in pushing ECP5 to an 800Mbps data rate because the DDR will be the bottleneck on my project.  I'm happy to implement and upstream a new PHY or modify as needed, with the caveat that I don't know much about the existing codebase (yet).