enjoy-digital/litedram #102: DFI Model: Add initialization capability

enjoy-digital, 2020-01-23:
Currently, it's not possible to initialize the memory content of the DFI model at startup. Being able to initialize it at startup (similar to what we can already do for inferred memories or block rams), would allow simulating SoCs with firmware/OS binaries stored in the SDRAM and directly jump to it at startup from the bios, avoiding serialboot or tftp that are slow in simulation.

An `init` parameter should be added to `SDRAMPHYModel`, taking a list with the initialization datas (simular to Migen's `Memory`). Since the SDRAM is organized in banks, the initialization data then needs to be spread among banks to initialize each bank memory: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/model.py#L49

enjoy-digital, 2020-01-27:
https://github.com/enjoy-digital/litedram/pull/104 implements the DFI model initialization, but there are still some limitations:

> for some reason I needed to reverse the order in each 4 32-bit word group. 

> Additionally it appears that at this point this won't work on memories with data bus wider than 32-bit because Migen seems to generate 32-bit wide init for 64-bit memories which then fails with $readmem file address beyond bounds of array

enjoy-digital, 2020-01-30:
@piotr-binkowski: sorry i haven't done tests yet on this, are you able to make progress on this? I just thought about a test that can be useful to verify the content:
````
lxsim --ram-init=<file.bin> (type escape during the boot to stay in the bios)
mr 0x40000000 <size>
````
and:
````
lxsim --with-sdram --sdram-init=<file.bin>
mr 0x40000000 <size>
````
should produce the same memory dump and could be useful to find initialization issues.

piotr-binkowski, 2020-01-30:
I made some progress and with my current code I can initialize memories wider than 32-bit but for widths above 128 it looks like order of 128-bit groups is also reversed. I tested it on 256 and 512-bit and in both cases order seen with `mr` was reversed compared to original file.
I also did some tests using DDR2/DDR3 modules but data ends up on wrong offsets.
I will create a WIP PR with that shortly

enjoy-digital, 2020-01-30:
When you are saying  `32/128/256/512-bit` is it the --sdram-data-width parameter you are passing or the controller's native data-width (which is `8*--sdram-data-width` parameter in DDR3 case and 4 phases). If that's with `--sdram-data-width`, supporting 128-bit is enough since the maximum tested on hardware is 72-bit (64-bit SO-DIMM  + 8-bit for ECC)

piotr-binkowski, 2020-01-30:
Yes, it is the `--sdram-data-width` parameter.
If 128-bit support is enough then I will now focus on getting init working on DDR modules

enjoy-digital, 2020-01-30:
128-bit is enough yes, no need to support more.

piotr-binkowski, 2020-01-31:
@enjoy-digital I did some tests to understand why I needed to reverse order of 32-bit words and it appears that data comes out of LiteDRAM in correct order and for some reason gets reversed in L2 Cache. I think that this might be caused by using `dsiplacer(...,reverse=True)` [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L541) and [here](https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L546) because after removing that option CPU sees the 32-bit words in correct order and still passes the memtest.

enjoy-digital, 2020-01-31:
@piotr-binkowski: thanks, this make sense. I just added a `l2_reverse` parameter to `SoCSDRAM`: https://github.com/enjoy-digital/litex/commit/eff9caee6a41f77f04f937faee65f84c095b6f0b. If the initialization in simulation are working fine with `l2_reverse=False`, we could use that and you could remove the reversing in the DFI model. Then we should probably make the `l2_reverse=False` the default behavior, but this will probably need some tests/adaptations on designs where SDRAM regions are used by the CPU and DMAs.

piotr-binkowski, 2020-02-03:
I tested this `l2_reverse` parameter and with it I can remove the order reversing code and still boot linux from SDRAM on linux-on-litex-vexriscv

PR #104 (2020-01-27): phy/model: add support for initializing memory from file
This PR adds support for initializing contents of `SDRAMPHYModel` which can greatly help with simulations.
This still needs some adjustments because LiteDRAM uses `ROW_BANK_COL` order and I initially went with simpler `BANK_ROW_COL` order. At this point data is visible in simulation but it is all over the place.

Closes #102
PR #124 (2020-02-03): phy/model: cleanup the memory init code
This PR removes code responsible for word order reversing as it is no longer needed since adding `l2_reverse` and additionally does a general cleanup by making a separate method from bank init code to only run it when needed

Closes #102
closed 2020-02-03 by enjoy-digital