enjoy-digital/litedram #103: LiteX BIOS fails to correctly set read delay, ECP5

gregdavill, 2020-01-24:
I've been running LiteDRAM on my OrangeCrab boards, this is a ECP5 platform using the CSFBGA285 package.

Out of 6 prototype r0.1 boards I had 2 fail the memtest in LiteX BIOS. 
Some further investigations determined that there were no shorts/connection issues. 
I've been poking around with the memory initialization code, and I can appear to get it working, but manually setting rd_dly values.

Adding the following snippet of C into the BIOS after the initialization results in passing memtest for 3 values of rd_dly.

```C
read_bitslip_rst(0);
	read_bitslip_rst(1);
	for(int j = 0; j < 8 ; j++){
		// Load a read delay for the current loop 
		read_delay_rst(0);
		read_delay_rst(1);
		for(int i=0;i<j;i++){
			read_delay_inc(0);
			read_delay_inc(1);
		}
		printf("## Testing with read_delay=%d ##\r\n", j);
		memtest();
		printf("\r\n");
	}
```

Here is a log from the serial
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|

 (c) Copyright 2012-2019 Enjoy-Digital

 BIOS built on Jan 23 2020 23:50:42
 BIOS CRC passed (6e0b3159)

 Migen git sha1: 16179b2
 LiteX git sha1: 5845df76

--=============== SoC ==================--
CPU:       VexRiscv @ 48MHz
ROM:       32KB
SRAM:      4KB
L2:        8KB
MAIN-RAM:  131072KB

--========== Initialization ============--
Initializing SDRAM...
SDRAM now under software control
Read leveling:
m0, b0: |11100000| delays: 01+-01
best: m0, b0 delays: 01+-01
m1, b0: |11100000| delays: 01+-01
best: m1, b0 delays: 01+-01
SDRAM now under hardware control
Memtest bus failed: 2/256 errors
Memtest data failed: 36/64 errors
Memtest addr failed: 8192/8192 errors
## Testing with read_delay=0 ##
Memtest OK
Memspeed Writes: 114Mbps Reads: 108Mbps

## Testing with read_delay=1 ##
Memtest OK
Memspeed Writes: 119Mbps Reads: 108Mbps

## Testing with read_delay=2 ##
Memtest OK
Memspeed Writes: 119Mbps Reads: 108Mbps

## Testing with read_delay=3 ##
Memtest bus failed: 3/256 errors
Memtest data failed: 29/64 errors
Memtest addr failed: 8192/8192 errors

## Testing with read_delay=4 ##
Memtest bus failed: 9/256 errors
Memtest data failed: 41/64 errors
Memtest addr failed: 1415/8192 errors

## Testing with read_delay=5 ##
Memtest bus failed: 7/256 errors
Memtest data failed: 9/64 errors
Memtest addr failed: 1706/8192 errors

## Testing with read_delay=6 ##
Memtest bus failed: 8/256 errors
Memtest data failed: 30/64 errors
Memtest addr failed: 1342/8192 errors

## Testing with read_delay=7 ##
Memtest bus failed: 9/256 errors
Memtest data failed: 21/64 errors
Memtest addr failed: 1756/8192 errors

--============== Boot ==================--
Booting from serial...
Press Q or ESC to abort boot completely.
sL5DdSMmkekro
Timeout
No boot medium found

--============= Console ================--
litex>
```

You can see the initialization correctly identifies that values of [0,1,2] are working, but for some reason it's not correctly locking that value in to the LiteDRAM core.

whatnick, 2020-01-24:
I can note that in my case delay=7 passes tests as well. However the runtime determined delay is not persisted, future memtests still fail.

gregdavill, 2020-01-25:
It appears that strobing the dly_sel lines after performing read_levelling, is all that's needed.
I'm not sure exactly why this is the case and why on some of my boards this wasn't required.

I suspect that strobing the DQSBUFM.PAUSE once all the READCLKSEL (rd_dly) values are set up gets everything in sync.

enjoy-digital, 2020-01-26:
Fixed with https://github.com/enjoy-digital/litex/pull/359.

closed 2020-01-26 by enjoy-digital