enjoy-digital/litedram #110: Benchmark: add benchmark script to Travis-CI

enjoy-digital, 2020-01-28:
Requires https://github.com/enjoy-digital/litedram/issues/109

The test should fail if any errors is detected during the test of the various configurations.

The results of the benchmark scripts should be stored somewhere to allow finding potential regressions in efficiency/performance on LiteDRAM.

jedrzejboczar, 2020-01-31:
With https://github.com/enjoy-digital/litedram/pull/123 merged, we could store the following artifacts:
* benchmark text summary (simple output redirection)
* plots
*  use `--output-cache` to store pickled pairs (benchmark_configuration, benchmark_output)

The pickled data can later be loaded by `run_benchmarks.py` without actually running the benchmarks, and so we can use it to generate output in different format if needed or to analyze LiteDRAM performance over time. 

If using `pickle`'s binary format is not ok, we can e.g. just store configurations and outputs in separate files, but pickling seems to be the easy solution.

mithro, 2020-01-31:
JSON seems like a better format than pickle?

enjoy-digital, 2020-01-31:
+1 for JSON.

closed 2020-02-18 by enjoy-digital