enjoy-digital/litedram #126: DFI Model: Add timing violation detection

piotr-binkowski, 2020-02-03:
Currently `SDRAMPHYModel` doesn't detect any timing violation issues, I think that we could enhance  it to verify if the DFI control signals are generated in a way that doesn't violate the timings for a selected SDRAM module.
From what I see the memory controller operates on clock cycles to calculate required timings so this approach should also be adopted by the PHY model instead of using raw `ns` values.
I'm also wondering how we should communicate to the user the fact that violation occured. Maybe initially we could create some signals that are asserted when violation occurs which can then be checked manually in the `.vcd` file. In the end we should probably find a better way to communicate the violations to users/CI system

@enjoy-digital please take a look and check if that makes sense

enjoy-digital, 2020-02-03:
@piotr-binkowski: yes this makes sense, just that i think it's better for the PHY model to work on clock cycles and then convert measurements to `ns` (knowing the clock period and number of phases, it should be a simple multiplication + addition) to do the check against the real `ns` values. This way the test will also cover eventual issues in the `ns` to `cycles` functions of the SDRAM Module.

For the export of the results, something similar to what is used for the benchmark could be used: https://github.com/enjoy-digital/litedram/blob/master/test/benchmark.py#L92-L99, the issues could be reported as soon they appears (useful when debugging) or summary at the end the simulation (useful for CI).

piotr-binkowski, 2020-02-14:
Initial support for this was added in #138
One thing missing currently that I'm aware of is support for postponing/pulling-in refresh cycles allowed by some DDR chips (DDR4, maybe other types)
@enjoy-digital is there anything else that you think should be added to the timing verifier block?

enjoy-digital, 2020-02-14:
@piotr-binkowski i will review more closely the timing checker and will answer here if i see missing features.

enjoy-digital, 2020-04-13:
The code has been reviewed and modified a little bit to improve readability.

closed 2020-04-13 by enjoy-digital