enjoy-digital/litedram #130: ECP5: CL settings behaviour

gregdavill, 2020-02-04:
I've populated one of my OrangeCrab r0.2 boards with this part: MT41K512M16HA-125:A (https://www.digikey.com.au/product-detail/en/alliance-memory-inc/MT41K512M16HA-125:A/1450-MT41K512M16HA-125:A-ND/10821776) to test out the extra address lines I'd added to support 8G DDR3.
I've created a custom Module based off the MT41K256M16, that seems to work fine.

I'm testing over the wishbone bridge, using a slightly modified version of this code: https://github.com/gregdavill/OrangeCrab/blob/master/gateware/litex_ddr3_test/test/test_sdram.py

What I'm finding is a bit weird. 
The gateware works great with the smaller 1G parts `MT41K64M16`. But when using the larger 8G part `MT41K512M16` I get errors with all the rdly settings.

However what I've just discovered is if I don't change anything with the gateware, but alter the CL setting on the DDR3 in the init sequence, from 6 to 7 it works as expected.

```
    ("Load Mode Register 0, CL=6, BL=8", 0x320, 0, dfii_command_ras|dfii_command_cas|dfii_command_we|dfii_command_cs, 200), # Autogenerated
    ("Load Mode Register 0, CL=7, BL=8", 0x330, 0, dfii_command_ras|dfii_command_cas|dfii_command_we|dfii_command_cs, 200), # Altered to CL=7
```

Do you have any insights as to why this could be happening?
I'll keep doing more testing on my hardware.

gregdavill, 2020-02-04:
I'm testing gateware on two different boards I have here.
- ECP5-25F + MT41K64M16
- ECP5G-85F + MT41K512M16

(I don't anticipate the issue is related to a different ECP5 part, but I can assemble another board to test that also.)

These are both generated with a clean LiteDRAM master 
```
 > MT41K64M16

m0, b0: |11100000| delays: 01+-01
best: m0, b0 delays: 01+-01
m1, b0: |11100000| delays: 01+-01
best: m1, b0 delays: 01+-01
```

```
 > MT41K512M16

m0, b0: |00000000| delays: 00+-00
best: m0, b0 delays: 00+-00
m1, b0: |00000000| delays: 00+-00
best: m1, b0 delays: 00+-00

### Proceeds to fail memtest ###
```

But here I've set a value of CL=7 into MR0.
```
 > MT41K64M16

m0, b0: |00001110| delays: 05+-01
best: m0, b0 delays: 05+-01
m1, b0: |00001110| delays: 05+-01
best: m1, b0 delays: 05+-01
```

```
 > MT41K512M16

m0, b0: |00011100| delays: 04+-01
best: m0, b0 delays: 04+-01
m1, b0: |00011100| delays: 04+-01
best: m1, b0 delays: 04+-01
```

enjoy-digital, 2020-02-04:
@gregdavill: i think the issue is that the read leveling has only been validated on a few boards and is not robust enough.

For you board, with the LiteX/LiteDRAM master, it seems the data are sampled too late and using `CL=7` (vs `CL=6`) adds one SDRAM clk (1/2 `sys_clk`) of delay to the read-datas. Can you try to see using `CL=6` and moving `dqs_read` one `sys_clk` before (`dqs_read.eq(rddata_ens[cl_sys_latency | rddata_ens[cl_sys_latency+1]`) also makes it work? If so, maybe we should add a register to configure the `dqs_read` sampling and use it in the software calibration to have 16 delays instead of 8.

gregdavill, 2020-02-04:
I think that adding the ability to move clock cycles is a good idea, to give a larger range to the read-levelling. I suppose that can't do any harm.

I've done some more tests, this is using a vanilla LiteDRAM.

The board fails at 48MHz. 
But passes when running at 96MHz, with delay: 02+-01. 

So some of this might simply be due to running the 8G part at such a slow speed.

enjoy-digital, 2020-02-04:
If you have time, i would be interested to know if doing the` dqs_read ` one cycle earlier works at 48MHz. That would mean moving it in software would make things work at both 48MHz and 96MHz.

daveshah1, 2020-02-04:
Low frequencies on the ECP5 do seem to be problematic in general - there has been unreliability observed on TrellisBoards (2x4Gb parts) at 55MHz or lower

gregdavill, 2020-02-05:
With CL=6, just changing `dqs_read.eq(rddata_ens[cl_sys_latency | rddata_ens[cl_sys_latency+1]` doesn't seem to work.

However changing cl, does work! (We have to +1 in the phy settings, so that we are still telling the RAM CL=6)
```
cl, cwl         = get_cl_cw(memtype, tck)
cl = cl - 1
...
  wrcmdphase    = wrcmdphase,
  cl            = cl + 1,
  cwl           = cwl,
```

So I just need to trace out what this effects and how we can make it a runtime adjustable value.

@daveshah1 good to know that I'm not the only one who has seen this. It would be nice if the core could support running at a lower frequency. When you say 'unreliability' Do you mean just getting things working? or longer term?

daveshah1, 2020-02-05:
By unreliability, I mean occasional memtest failures on about 20% of boards. Never seen unreliability if memtest passes.

closed 2021-01-12 by enjoy-digital