enjoy-digital/litedram #133: Benchmark: BIST not generating random address

jedrzejboczar, 2020-02-05:
`random` register in `_LiteDRAMBISTGenerator` and `_LiteDRAMBISTChecker` is used for random data generation, but address is always sequential:
https://github.com/enjoy-digital/litedram/blob/master/litedram/frontend/bist.py#L145

Random data has no impact on LiteDRAM performance, shouldn't we generate random address instead? Or add separate registers for random data and random address?

enjoy-digital, 2020-02-06:
`random` was used to test integrity of the data, this has indeed no impact on the efficiency/performance. We could add two separate CSR: `random_data` and `random_addr`. But we should probably add another CSR to limit the range of the random address, so maybe add a `end` CSR (similar to `base)` and random addressing would be done between `base` and `end.`

jedrzejboczar, 2020-02-07:
Is there any good way of limiting the generated value to a range when `end` is a CSR? Because I cannot think of a nice solution.

It would be the easiest if `base` and `end` were just parameters passed to `__init__`, even better if we only allowed `end - base` to be a power of 2. Or just pass `range_bits`. We could then just create an LSFR with desired output width.

When `end` is a CSR with arbitrary address, we may check if `generated > end` and if not, then choose some other value. This could be just `base`/`end`, or we could switch to a simple counter for as long as we get random values out of range. But this will give very bad results for small ranges, because we would then almost always generate values out of range, so almost all values would be from the fallback strategy.

enjoy-digital, 2020-02-08:
In fact we need `base` and `end` to be dynamic (so CSR) for the hardware tests were we will want to re-configure `_LiteDRAMBISTGenerator` and `_LiteDRAMBISTGenerator:`

We can do the assumption that the configured range will always be a power of 2, and then you can probably do something similar to:

```
addr_gen = Generator(31, n_state=31, taps=[27, 30]) # PRBS31
addr_mask = Signal(32)
self.comb += addr_mask.eq(self.end.storage - self.base.storage)
self.comb += dma.sink.address.eq(self.base.storage + addr_gen.o & addr_mask)
```

jedrzejboczar, 2020-02-10:
Thanks, that's nice and simple, I'll implement it this way.

closed 2020-02-12 by enjoy-digital