enjoy-digital/litedram #145: init_done never goes to 1 in simulation

kessam, 2020-02-17:
I have generated the Verilog core of litedram and connected it to a simple AXI master. Then, I ran simulation using vsim on such system for ~100ms (1,000,000 clock cycles).  
It's found that init_done signal is always 0:
[image]

So the question is, how long should I wait till init_done signal goes to 1. I know that I should wait some time till the CPU does the initialization, but I think 100ms is enough time to do that.

Here is the configuration I used to generate the core:
`{
    # General ------------------------------------------------------------------
    "cpu":        "vexriscv",  # Type of CPU used for init/calib (vexriscv, lm32)
    "speedgrade": -1,          # FPGA speedgrade
    "memtype":    "DDR3",      # DRAM type

    # PHY ----------------------------------------------------------------------
    "cmd_delay":       0,             # Command additional delay (in taps)
    "cmd_latency":     0,             # Command additional latency
    "sdram_module":    "MT41J256M16", # SDRAM modules of the board or SO-DIMM
    "sdram_module_nb": 4,             # Number of byte groups
    "sdram_rank_nb":   1,             # Number of ranks
    "sdram_phy":       "V7DDRPHY",    # Type of FPGA PHY

    # Electrical ---------------------------------------------------------------
    "rtt_nom": "60ohm", # Nominal termination
    "rtt_wr":  "60ohm", # Write termination
    "ron":     "34ohm", # Output driver impedance

    # Frequency ----------------------------------------------------------------
    "input_clk_freq":   100e6, # Input clock frequency
    "sys_clk_freq":     50e6, # System clock frequency (DDR_clk = 4 x sys_clk)
    "iodelay_clk_freq": 200e6, # IODELAYs reference clock frequency

    # Core ---------------------------------------------------------------------
    "cmd_buffer_depth": 16,    # Depth of the command buffer

    # User Ports ---------------------------------------------------------------
    "user_ports_nb":       1,     # Number of user ports
    "user_ports_type":     "axi", # Type of ports (axi, wishbone, native)
    "user_ports_id_width": 1,     # AXI identifier width
   
    # CSR Port -----------------------------------------------------------------
    "csr_expose": "False", # Expose CSR bus as I/Os
    "csr_align" : 32,      # CSR alignment
}
`

kessam, 2020-02-17:
I want also to ask, is there something missing in the simulation setup. I've read in other discussions that I need to boot some code on the CPU to initialize the controller. How can this be done?

enjoy-digital, 2020-02-17:
The generator is generating a `litedram_core.init` file that contains the code executed by the CPU. Are you providing this file to the simulator? Do you see some activities on the CPU buses?

kessam, 2020-02-17:
Hi, 
Thanks for this very quick response :)
Yes I provided the memory file to the simulator. I attached a screen shot of the bus and memory activities:
[image]

Thanks,
Khaled

enjoy-digital, 2020-02-17:
The simulation will probably produce errors during the leveling and memtest, you could disable it for simulation by commenting out:
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/sdram.c#L979
and
https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/bios/sdram.c#L985-L991

And then re-generate the core.

With that, you should see the init_done set to 1.

kessam, 2020-02-17:
Unfortunately, this didn't work. init_done is still 0.
I attached a compressed file for my VCD. Please extract it and open with gtkwave.
[issue_145.vcd.zip](https://github.com/enjoy-digital/litedram/files/4216110/issue_145.vcd.zip)

kessam, 2020-02-19:
@enjoy-digital 
Finally, init_done became 1, but after connecting the litedram to a simulation model of DDR3. I used the DDR3 simulation model shipped by Xilinx MIG example. 

P.S I had to run the simulation for 25ms (about 2.5 hours on my machine :) ) to get the init_done to 1

enjoy-digital, 2020-02-20:
@kessam: ok good. The UART is probably slowing that your simulation a lot, if you want to speed this up, you can add `uart_name="stub"` to https://github.com/enjoy-digital/litedram/blob/master/litedram/gen.py#L339.

enjoy-digital, 2020-03-04:
We can close since the question has been answered.

closed 2020-03-04 by enjoy-digital

Yuxin-Yu, 2023-06-13:
Hello, bro,can you tell me whether your axi write port is normal? I also used the xilinx mig example to replace the lig ram with the mig, and found that the write function is abnormal.Which like:https://github.com/enjoy-digital/litedram/issues/342
@kessam