enjoy-digital/litedram #176: Vivado constraints are not applied

kessam, 2020-04-05:
The following Vivado constraints are not applied:
`set_false_path -quiet -to [get_nets -quiet -filter {mr_ff == TRUE}]`
`set_false_path -quiet -to [get_pins -quiet -filter {REF_PIN_NAME == PRE} -of [get_cells -quiet -filter {ars_ff1 == TRUE || ars_ff2 == TRUE}]]`
`set_max_delay 2 -quiet -from [get_pins -quiet -filter {REF_PIN_NAME == Q} -of [get_cells -quiet -filter {ars_ff1 == TRUE}]] -to [get_pins -quiet -filter {REF_PIN_NAME == D} -of [get_cells -quiet -filter {ars_ff2 == TRUE}]]
`

For the first one, 'set_false_path' doesn't applied to a 'Net' destination. Instead, option '-through' should be used for nets:
`set_false_path -through  [get_nets -hierarchical  -filter {mr_ff == TRUE}]`

And, '-hierarchical' is missing from 'get_cells' command, so they should be:
`set_false_path -to [get_pins -filter {REF_PIN_NAME == PRE} -of [get_cells -hierarchical -filter {ars_ff1 == TRUE || ars_ff2 == TRUE}]]`

Furthermore, set_max_delay cannot have a 'Q pin' starting point. Instead, it should be 'C pin':
`set_max_delay 2 -from [get_pins -filter {REF_PIN_NAME == C} -of [get_cells -hierarchical  -filter {ars_ff1 == TRUE}]] -to [get_pins -filter {REF_PIN_NAME == D} -of [get_cells -hierarchical  -filter {ars_ff2 == TRUE}]]`

enjoy-digital, 2020-04-05:
Thanks for reporting and investigating. Do you see warnings reported by Vivado with the upstream commands? Would you mind creating a PR with the fixes you identified? Thanks.

kessam, 2020-04-05:
Yes I do see WARNINGS/CRITICAL WARNINGS when removing the '-quiet' option. I've created a Pull Request of the proposed fixes. Let me know your thoughts.

enjoy-digital, 2020-04-06:
Thanks for https://github.com/enjoy-digital/litex/pull/448, it has been merged. Constraints were probably working when introduced but having the `quiet` enabled prevented detecting the regression when introduced.

closed 2020-04-06 by enjoy-digital