enjoy-digital/litedram #182: Refresh S6DDRPHY

enjoy-digital, 2020-04-23:
The Spartan6 S6DDRPHY needs to be refreshed with the simplifications/improvements made to the others PHYs.

cklarhorst, 2020-07-16:
@enjoy-digital 
Could you please elaborate a bit more on what simplifications/improvements are not implemented/missing for S6DDRPHY?
Is one of the missing features that S6DDRPHY doesn't use IDELAY/ODELAY for data alignment?

enjoy-digital, 2020-07-31:
@cklarhorst: sorry i forgot to answer, the S6DDRPHY is the first PHY that has been created and we improved things since then, so the architecture would need to be updated a bit and it's also possible some features are missing (DQS Pre/Post amble for example). That's not necessarilly something easy to work on when not familiar with LiteDRAM, so i was planning to work on that, but if you want to have a closer look, i could help you getting started.

norbertthiel, 2021-02-10:
@enjoy-digital 
I am currently implementing Litex on a custom XC6SLX16-3 board which comes with LPDDR (Winbond W949D6KBH)
It works fine using parameters copy from the pipistrello board - CPU clock at 83.3 MHz and memory at 166MHz
* trying 200MHz memory clock and 100MHz CPU clock results in mem_test failing 

So, before start digging into this, I am wondering if you started refreshing the S6DDRPHY already?

enjoy-digital, 2021-02-11:
@norbertthiel: Hi, the work has not been initiated in this no.

norbertthiel, 2021-02-11:
> @norbertthiel: Hi, the work has not been initiated in this no.

Hi and thanks for the reply. Any short-term plans for initiating this?

Maybe you can point me were to start looking at to fix my "problem":
* 83,3MHz sys, 166MHz sdram_half, 333MHz sdram_full (pippistrello org settings)  => works
* 90Mhz sys, 180Mhz sdram_half, 360MHz sdram_full => when using "mem_write 0x40000000 0x01020304 8" I get all 8 values written = but at an offset of roughly 124 bytes there is a garbage word written to memory (3 garbage words written at offset 0x8e, 0x9e, 0x116) ... I have no idea were this is coming from  - any idea/hint?

memory supports 200MHz, FPGA has speedgrade 3

norbertthiel, 2021-02-11:
@enjoy-digital 
like this:
[image]

enjoy-digital, 2021-02-11:
Hi @norbertthiel,

the S6DDRPHY needs to be refreshed since as you saw is not very flexible and can require manual adjustments to get it working. Are the timings OK at 100MHz sys_clk? If yes, you could try to adjust the passed `rd_bitslip/wr_bitslip` values: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s6ddrphy.py#L38.

I'd like the refresh work to be done in the next months but can't guarantee it.

Florent

norbertthiel, 2021-02-11:
> Hi @norbertthiel,
>
> the S6DDRPHY needs to be refreshed since as you saw is not very flexible and can require manual adjustments to get it working. Are the timings OK at 100MHz sys_clk? If yes, you [...]

Thanks Florent, yepp I saw that a few timings are not flexible and I am going to play with that. Looking forward to your refresh work. Any other sys_clk timing I tried did not work.

I will play with bitslip and also thought that read_latency could have an impact.

> Are timings OK at 100MHz sys_clk?

Didn’t connect litescope yet -probably gonna try soon - or do you suggest to use simulation?

enjoy-digital, 2021-02-11:
For the timings, this is reported by ISE.

norbertthiel, 2021-02-11:
> For the timings, this is reported by ISE.

Ah those, sorry, yepp, constraints met, no problems reported

norbertthiel, 2021-02-23:
@enjoy-digital 
quick update: sys_clk 100MHz (400MHz sdram_full) works with:
* clk_sdram_half_shifted - phase shift 240
* cd_sdram_half phase shift 270
* rd_bitslip: 0
* wr_bitslip: 3
* dqs_ddr_alignment: C1
[image]

enjoy-digital, 2021-02-25:
Great @norbertthiel, thanks for the feedback!

norbertthiel, 2021-02-25:
> Great @norbertthiel, thanks for the feedback!

in fact phase shifts 230 and 260 work stable - 240 and 270 did occasionally result in memtest errors - 260 and 230 seems to work reliably

and let me just say how much I enjoy using LiteX

enjoy-digital, 2021-02-25:
Thanks, tweaking the S6DDRPHY's settings is not necessarily the easiest thing to start with :) (in 7-Series / Ultrascale these manual settings are almost all automated and the aim of this issue is also simplify this for Spartan6), but glad you manage to get it working, things should be easier now.

enjoy-digital, 2026-11-09:
The S6DDRPHY has been cleaned up and documented in #480 (no functional change, generated Verilog unchanged).

The remaining work (runtime read/write leveling with ISERDES2 bitslip/IODELAY2 CSRs, configurable CL/CWL, DQS preamble) would be a substantial rework that requires Spartan-6 hardware to validate. Since Spartan-6 is a legacy family and the PHY works with the manual settings (see the parameters documented in #480 and the working 100MHz settings above), we don't plan to work on it. Contributions are still welcome if someone with hardware wants to tackle it; we'd be happy to review.

closed 2026-11-09 by enjoy-digital