enjoy-digital/litedram #184: Inconsistencies between SPD data and datasheets.

enjoy-digital, 2020-04-25:
From https://github.com/enjoy-digital/litedram/pull/181:

For some modules there seems to be inconsistencies between `tRCD` and `tRP` values in the SPD data, in the SO-DIMM module datasheet and in the datasheet of the chip used in the module. The values from SPD data were used when existing for the given module.

enjoy-digital, 2020-04-25:
@jedrzejboczar: Can you provide a module where you saw an inconsistency?

jedrzejboczar, 2020-04-27:
When looking for `tRCD`, `tRP` and `tRC` for our modules both SPD data and values in the module datasheet are the same, but it seems to me that those values differ from what is written in the datasheets of the actual chips used in the modules. Or maybe I am interpreting the datasheets incorrectly. Anyway it is probably the most reasonable to use what is in SPD and module datasheet.

Below I've written down the values extracted from the datasheets. The relevant datasheets are attached in [datasheets.zip](https://github.com/enjoy-digital/litedram/files/4539094/datasheets.zip).

```
MT8JTF12864 (1333):
* module datasheet: 
    * tRCD, tRP, tRC: 13.125, 13.125, 49.125
* chip datasheet: MT41J128M8, 1333, 9-9-9
    * Table 52.
        * tRCD, tRP, tRC: 13.5, 13.5, 49.5
* spd data:
    * tRCD, tRP, tRC: 13.125, 13.125, 49.125
```

Here, the chip datasheet specifies 13.5. The value of 13.125 is specified only for 7-7-7, but then tRC would be 50.625.

The following modules all use the same chip MT41K512M8 and there is also a mismatch between chip datasheet and module/SPD.

```
MT8KTF51264:
* module datasheet: 
    * (1G4, 1333) tRCD, tRP, tRC: 13.125, 13.125, 49.125
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125
    * (1G9, 1800) tRCD, tRP, tRC: 13.125, 13.125, 47.125
* chip datasheet:
    * Table 52.
        * (1G4, 1333 -> MT41K512M8,    9-9-9) tRCD, tRP, tRC: 13.5,  13.5,  49.5
        * (1G6, 1600 -> MT41K512M8, 11-11-11) tRCD, tRP, tRC: 13.75, 13.75, 48.75
        * (1G9, 1800 -> MT41K512M8, 13-13-13) tRCD, tRP, tRC: 13.91, 13.91, 47.91
* spd data:
    * (1G4, 1333) tRCD, tRP, tRC: 13.125, 13.125, 49.125
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125
    * (1G9, 1800) tRCD, tRP, tRC: 13.125, 13.125, 47.125

MT18KSF1G72HZ:
* module datasheet: 
    * (1G4, 1333) tRCD, tRP, tRC: 13.125, 13.125, 49.125
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125
* chip datasheet:
    * Table 52.
        * (1G4, 1333 -> MT41K512M8,    9-9-9) tRCD, tRP, tRC: 13.5,  13.5,  49.5
        * (1G6, 1600 -> MT41K512M8, 11-11-11) tRCD, tRP, tRC: 13.75, 13.75, 48.75
* spd data:
    * (1G4, 1333) tRCD, tRP, tRC: 13.125, 13.125, 49.125
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125

MT16KTF1G64HZ:
* module datasheet: 
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125
    * (1G9, 1800) tRCD, tRP, tRC: 13.125, 13.125, 47.125
* chip datasheet:
    * Table 52.
        * (1G6, 1600 -> MT41K512M8, 11-11-11) tRCD, tRP, tRC: 13.75, 13.75, 48.75
        * (1G9, 1800 -> MT41K512M8, 13-13-13) tRCD, tRP, tRC: 13.91, 13.91, 47.91
* spd data:
    * (1G6, 1600) tRCD, tRP, tRC: 13.125, 13.125, 48.125
    * (1G9, 1800) tRCD, tRP, tRC: 13.125, 13.125, 47.125
```

enjoy-digital, 2026-11-09:
Coming back to this: for a module, the SPD data (which matches the module datasheet here) is what the module vendor guarantees for the chips it fits, while the chip datasheet gives the limits of a speed bin, which tested chips can beat. So the SPD/module datasheet values are the right ones for the module definitions, and that's what `modules.py` uses (13.125 ns for these modules, checked against the SPD dumps in `test/spd_data` by `test_modules`).

It also doesn't change the controller timings: with the rounding to sys_clk cycles, 13.125, 13.5, 13.75 and 13.91 ns give the same tRCD/tRP at 1:2 and 1:4 from 50 to 233 MHz sys_clk.

So no change needed, closing.