enjoy-digital/litedram #194: ECP5: Use ODDRX1F for address and command bus

kingoflolz, 2020-05-15:
It should be possible to use ODDRX1F instead of ODDRX2F for the address and command bus, so that these signals can be placed on the top IO bank instead of just the left and right.

This seems like a relatively straightforward change, and I would be happy to provide a pull request.

enjoy-digital, 2020-05-19:
Hi @kingoflolz, if you do the change and confirm the Versa ECP5 target (https://github.com/enjoy-digital/litex/blob/master/litex/boards/targets/versa_ecp5.py) is still building fine with it, that's probably something we could merge yes.

enjoy-digital, 2020-06-22:
@kingoflolz: sorry i won't be able to spend time investigating this and i'm not sure it will be the case for someone else in the near future. I'll however be happy to test and integrate a validated solution if provided.

madscientist159, 2021-05-30:
@enjoy-digital Since it's been a year, and we've run straight into this problem on some custom hardware, any chance of re-visiting this request?

enjoy-digital, 2021-05-30:
@madscientist159: If I remember correctly replacing the ODDRX2F with ODDRX1F was not as straightforward as expected and some time would need to be spent understanding why. If we want to go further on this, I would happy to test and do the integration of a validated solution (or spend time on this is someone has funds for it).

madscientist159, 2021-06-02:
We've investigated the issue and come up with a working solution here, tested to work on the Versa board:
https://gitlab.raptorengineering.com/kestrel-collaboration/kestrel-litex/litedram/-/tree/ecp5ddr1x

The problem wasn't clock skew between `ECLK`/`SCLK`, it was the phase shift introduced by the `ODDRX2` primitive configured the way it was in LiteX.  The `ODDRX1` primitive has less effective latency when clocked from `SCLK`, and its output needed to be delayed by one `ECLK` period as a result.

It isn't particularly easy to generate a low-skew inverted clock signal (needed to drive the `ODDRX1` primitive) from an existing edge clock on these devices.  We settled on a second `CLKDIVF` with some logic to use the `ALIGNWD` signal to ensure the clock is inverted, but the obvious disadvantage of this approach is the need to allow the signals to cross to the other side of the device (wasting `ECLK` resources).

I do note that Lattice's documentation encourages the use of the TOP I/O bank for address and command lines.  As a result I think it is important that the option be available to use the device as Lattice specifies, even if the logic for that case is a bit more complex.

enjoy-digital, 2021-06-08:
Thanks a lot @madscientist159 for the investigation and proposed solution, that's exactly the analysis that was required. The approach looks good yes and creating a PR would simplify integrating your solution yes. We can probably simplify the code a bit by creating wrappers for the primitives and avoid some python code duplication but I'm happy to handle this part.

aacuevas, 2022-02-18:
Hi. I am in such a case, in which we require to move the address lines to a `TOP` bank, so here is some feedback.
Our pinout requires data and command lines connected to side bank 7 and address to top bank 0, so they share a corner. We are also using Diamond due to some requisites, which was easily addressed.

When trying to build the design from @madscientist159, though, the router gave us an error related to DDR placement.

We have solved this by removing both `ECLKBRIDGECS` and connecting `sys_2x_i_clk` directly to the two `ECLKSYNCB` modules that feed the edge clocks to both `CLKDIVF' for `sys_clk` and @madscientist159's phase-inverted `sys_inv_clk`. This builds for our required pinout.

We have tested these changes with an OrangeCrab (Versas are not available) and they work. In this board, both data and address are connected to two side banks, so there might be unforeseen consequences of moving address to the top bank, but a design with that pinout pases PAR and the logic seems to work, so we are hopeful.

I will write an update once we build our custom hardware and test the final configuration, in case it might be useful for anyone else.

kkanics, 2024-06-20:
Any updates on integrating this?

enjoy-digital, 2026-11-09:
Hi,

sorry for the very long delay. Following the analysis from @madscientist159 and the feedback from @aacuevas, #477 adds a `cmd_gearing="x1"` option to `ECP5DDRPHY` (`cmd_gearing: x1` in `litedram_gen`) that drives CK/address/command with `ODDRX1F`, so they can be placed on the top banks.

Compared with the 2021 branch, it doesn't need the inverted `sys` clock (second `CLKDIVF`, `ALIGNWD` logic and `ECLKBRIDGECS`): the commands are delayed by one phase with a register instead, which gives the same output as `ODDRX2F` in the Lattice simulation models. CK comes from a PLL output phase-shifted by `ck_phase` (default 45°): taken straight from the PLL, CK comes out early compared with the commands and DQS, which go through the `ECLKSYNCB`.

Tested on an ECPIX-5 with its standard pinout (CK/address/command on a side bank): BIOS read leveling and memtest pass from DDR3-300 to DDR3-600, at 1:2 and 1:4, like the default `ODDRX2F`. With CK/address/command moved to top bank 0, nextpnr places and routes the design, but the board isn't wired that way, so this part hasn't been tested on hardware.

If you have hardware with CK/address/command on a top bank (@aacuevas, @kkanics), testing #477 would help a lot. The best `ck_phase` may differ there, so please report the values you try.

enjoy-digital, 2026-11-09:
#477 is now merged: `cmd_gearing="x1"` (`cmd_gearing: x1` and `ck_phase` in `litedram_gen`) is available on master. The default stays `x2` (`ODDRX2F`): `x1` needs an extra phase-shifted CK clock from the CRG, and is only useful when CK/address/command are on a top bank.

Since the feature requested here is implemented, I'm closing this. If you test it on hardware with CK/address/command on a top bank, please open a new issue with your results (working or not, and the `ck_phase` you used).

closed 2026-11-09 by enjoy-digital