enjoy-digital/litedram #208: Bad address alignment in ECP5DDRPHY (maybe others?)

kekiefer, 2020-07-19:
Recent changes made for #206 (https://github.com/enjoy-digital/litedram/commit/2c5fc6621a1087c2da2703d736f0ded6f26dbce3#diff-e10627e5a3f59b1f66ff980672b3f450R48) have broken DDR3 support in ECP5DDRPHY. At risk of unintentionally breaking stuff for hardware I don't have, I thought it best to raise a flag in case someone can advise as to the right way to handle this.

enjoy-digital, 2020-07-19:
Thanks @kekiefer for reporting, this has been fixed with https://github.com/enjoy-digital/litedram/commit/f51052f8b737156d1e257eff7cd3259cb56d0d1b and tested on OrangeCrab. Could you confirm it's also working on your side?

kekiefer, 2020-07-19:
That did the trick, thanks!

closed 2020-07-19 by kekiefer