enjoy-digital/litedram #212: Check DM polarity on DDR4

enjoy-digital, 2020-07-28:
From #192:

> reading through the datasheet of MT40A256M16 it is written that the data mask feature is controlled by MR5[10], so in our case it is disabled. It also seems that data mask is inverted, i.e. high DM means that given byte is being written and low DM means that given byte gets masked and ignored ("WRITE Burst Operation", p. 225). I tried enabling data mask in MR5 and tried running with both inverted and not inverted DM, but with no success.

jedrzejboczar, 2020-11-16:
I encountered the same issue on ZCU104 after disabling L2 cache. Previously the memory test must have been passing because, with L2 cache, there was no need for masking write data. I was able to fix the issue by inverting the signal in `USDDRPHY` and enabling the configuration bit in MR5[10].

Do you think that this should be done in PHY or rather in the MC itself (as described here: http://www.ddr-phy.org/forum/topics/dfi-wrdata-mask-polarity-for-ddr4-dfi-3-0-onwards)? If so it should probably be done in the crossbar? I can then create a PR with the fixes.
 
I had one more issue on ZCU104 - the Cmd/Clk delays and write DQ/DQS delays were not being incremented. Incrementing worked when manually writing to the CSRs from CLI. I was able to fix that by adding `cdelay(100)` after some of the `*_inc_write()` calls. I will create a PR to litex with these fixes, but I was workling on the versions from https://github.com/antmicro/litedram/tree/row-hammer and https://github.com/antmicro/litex/tree/row-hammer, so I will have to rebase.

enjoy-digital, 2020-11-16:
Thanks, i'll have a closer look tomorrow. For the delays, i also had a similar  issue on some build/boards and that's also what i was trying to solve with https://github.com/enjoy-digital/litex/commit/9fe3a42072519131d3a1c4e7dfb49b033fa28b8b. I'll have a look at the changes you did on that.

enjoy-digital, 2020-11-17:
@jedrzejboczar: I also think inverting in the PHY is the right way to do it: this way the DFI signals have the same meaning internally (even if that's not what suggest the answer in the forum). I merged your implementation with: https://github.com/enjoy-digital/litedram/commit/6b9f1bd0d863acdfe32ff9d0a05a9af2e1694f6b.

I'm now going to look at the delays to compare my changes and yours.

enjoy-digital, 2020-11-17:
You added additional delays that i just merged with https://github.com/enjoy-digital/litex/commit/3bd5d6cc0ee2daaa970c9f6bba2ae1c6c095c12f.

closed 2020-11-17 by enjoy-digital