enjoy-digital/litedram #213: Arty A7 LiteX SoC memory test runs extremely slowly

alanvgreen, 2020-07-30:
I am finding the LiteX memory test runs extremely slowly - about 1 # per 5 minutes. Originally I thought it was hung as per https://github.com/enjoy-digital/litex/issues/609, but after investigating found that reverting to an older version of litedram would avoid the problem.

The problem is not present at 71b991e but is present at 02e67ec7.

While experiencing this problem, I was not able to use UARTBone. Perhaps the Wishbone bus was being kept busy?

enjoy-digital, 2020-07-30:
@alanvgreen: that's indeed very slow and not to be expected. It's as if the bus timeout was occuring on each access. Here is a bitstream i just tested from the arty target in litex-boards: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/arty.py built with `./arty.py --build --load`, would you mind testing it? If working, could you give more information about the design you are testing? [arty_2020_07_30.zip](https://github.com/enjoy-digital/litedram/files/5000374/arty_2020_07_30.zip)

jedrzejboczar, 2020-07-30:
@alanvgreen From the commits you mentioned it is probably an issue either with wishbone2native or with the up-converter itself. Could you please check what wishbone and LiteDRAM port data widths you're using? You could just insert print statements in `LiteDRAMWishbone2Native`:
```python
class LiteDRAMWishbone2Native(Module):
    def __init__(self, wishbone, port, base_address=0x00000000):
        wishbone_data_width = len(wishbone.dat_w)
        port_data_width     = 2**int(log2(len(port.wdata.data))) # Round to lowest power 2

        print('Wishbone:', wishbone.data_width)
        print('LiteDRAM:', port.data_width)
```

alanvgreen, 2020-07-30:
Thank you both for the quick responses.

@enjoy-digital: The bitstream you provided memory tests at the usual speed.

@jedrzejboczar The output is:

```
Wishbone: 128
LiteDRAM: 128
```

I'm using lxbuildenv with a light wrapper around arty.py from litex_boards. Here is the build time output.

```
(arty) platforms@platforms-VirtualBox:~/arty/soc2 (master *+)$ ./soc2.py --build
lxbuildenv: v2020.6.1.1 (run ./soc2.py --lx-help for help)
INFO:SoC:        __   _ __      _  __  
INFO:SoC:       / /  (_) /____ | |/_/  
INFO:SoC:      / /__/ / __/ -_)>  <    
INFO:SoC:     /____/_/\__/\__/_/|_|  
INFO:SoC:  Build your hardware, easily!
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Creating SoC... (2020-07-31 05:59:37)
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:FPGA device : xc7a35ticsg324-1L.
INFO:SoC:System clock: 100.00MHz.
INFO:SoCBusHandler:Creating Bus Handler...
INFO:SoCBusHandler:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoCBusHandler:Adding reserved Bus Regions...
INFO:SoCBusHandler:Bus Handler created.
INFO:SoCCSRHandler:Creating CSR Handler...
INFO:SoCCSRHandler:8-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging (Up to 32 Locations).
INFO:SoCCSRHandler:Adding reserved CSRs...
INFO:SoCCSRHandler:CSR Handler created.
INFO:SoCIRQHandler:Creating IRQ Handler...
INFO:SoCIRQHandler:IRQ Handler (up to 32 Locations).
INFO:SoCIRQHandler:Adding reserved IRQs...
INFO:SoCIRQHandler:IRQ Handler created.
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:Initial SoC:
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoC:32-bit wishbone Bus, 4.0GiB Address Space.
INFO:SoC:8-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging (Up to 32 Locations).
INFO:SoC:IRQ Handler (up to 32 Locations).
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCCSRHandler:ctrl CSR allocated at Location 0.
INFO:SoCBusHandler:io0 Region added at Origin: 0x80000000, Size: 0x80000000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:cpu_bus0 added as Bus Master.
INFO:SoCBusHandler:cpu_bus1 added as Bus Master.
INFO:SoCCSRHandler:cpu CSR allocated at Location 1.
INFO:SoCBusHandler:rom Region added at Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:rom added as Bus Slave.
INFO:SoC:RAM rom added Origin: 0x00000000, Size: 0x00008000, Mode: R, Cached: True Linker: False.
INFO:SoCBusHandler:sram Region added at Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCBusHandler:sram added as Bus Slave.
INFO:SoC:RAM sram added Origin: 0x01000000, Size: 0x00002000, Mode: RW, Cached: True Linker: False.
INFO:SoCCSRHandler:identifier_mem CSR allocated at Location 2.
INFO:SoCCSRHandler:uart_phy CSR allocated at Location 3.
INFO:SoCCSRHandler:uart CSR allocated at Location 4.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCCSRHandler:timer0 CSR allocated at Location 5.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoCBusHandler:csr Region added at Origin: 0x82000000, Size: 0x00010000, Mode: RW, Cached: False Linker: False.
INFO:SoCBusHandler:csr added as Bus Slave.
INFO:SoCCSRHandler:bridge added as CSR Master.
INFO:S7PLL:Creating S7PLL, speedgrade -1.
INFO:S7PLL:Registering Single Ended ClkIn of 100.00MHz.
INFO:S7PLL:Creating ClkOut0 sys of 100.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut1 sys4x of 400.00MHz (+-10000.00ppm).
INFO:S7PLL:Creating ClkOut2 sys4x_dqs of 400.00MHz (+-10000.00ppm).
[... 50 lines cut ...]
- rom
- sram
- csr
- main_ram
- vexriscv_debug
INFO:SoC:8-bit CSR Bus, 32-bit Aligned, 16.0KiB Address Space, 2048B Paging (Up to 32 Locations).
CSR Locations: (11)
- ctrl           : 0
- cpu            : 1
- identifier_mem : 2
- uart_phy       : 3
- uart           : 4
- timer0         : 5
- ddrphy         : 6
- sdram          : 7
- leds           : 8
- i2c            : 9
- cam_leds       : 10
INFO:SoC:IRQ Handler (up to 32 Locations).
IRQ Locations: (2)
- uart   : 0
- timer0 : 1
INFO:SoC:--------------------------------------------------------------------------------
INFO:SoCBusHandler:Interconnect: InterconnectShared (3 <-> 5).
```

alanvgreen, 2020-07-30:
It seems to work with everything at head. I was using an older version of litex to avoid https://github.com/enjoy-digital/litex/issues/609 but with versions aligned, it seems to be behaving.

litex: 86e910d
litex_boards: 929e55d
litedram: 47a0d5f

Apologies for the noise.

closed 2020-07-30 by alanvgreen

enjoy-digital, 2020-07-31:
@alanvgreen, no problem. I'm looking at https://github.com/enjoy-digital/litex/issues/609, it will be fixed today.