enjoy-digital/litedram #215: Support for DRAM on Lattice ICE40HX devices?

adamfeuer, 2020-08-14:
Hi, is LiteDRAM known to work on Lattice ICE40HX devices? I didn't see an example for it... has anyone tried it yet?

daveshah1, 2020-08-14:
I believe it has been used successfully for SDRAM as part of a LiteX SoC. There is no iCE40 PHY for any of the DDR variants.

adamfeuer, 2020-08-14:
@daveshah1 Thanks. SDRAM would work for me. 

Does it make sense to create a DDR PHY for the the iCE40? If so, what would it take to do it? How were the existing DDR PHYs created and debugged (for instance the [s6ddrphy.py](https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s6ddrphy.py)?)

adamfeuer, 2020-08-15:
I see there's a DDR PHY for ECP5. That would also work for me. Thanks.

closed 2020-08-15 by adamfeuer