enjoy-digital/litedram #216: Check default DDR3 termination settings/power consumption on ECP5/OrangeCrab

enjoy-digital, 2020-08-24:
Investigation from @emeb posted on 1BitSquared Discord channel:

> A few weeks back @GregDavill gave some advice on reducing power dissipation in the memory interface. At the time I hadn't enough familiarity with Litex to make much use of that, but I've been learning so here's a bit of data I took. There were three main recommendations:
>1) Disable "virtual VCCIO" - these are basically IOBs that aren't balled-out but are tied to package VCCIO. Driving a '1' on then reduces the supply impedance to the on-chip power rails. These connections are made directly in the primary Litex source code for a design and are easy to simply comment out.
>2) Disable ODT. This requires editing the litedram/core/multiplexer.py file as mentioned below.
>3) Disable DQ termination. This is found in the OrangeCrab platform definition file where I changed Misc("TERMINATION=75") to Misc("TERMINATION=OFF").
>Measurements: This was done with the OrangeCrab mfg test software and a cheap eBay USB power meter.
>* Original draw : 0.29A
>* Disable Termination: 0.20A
>* Disable Virtual VCCIO : 0.17A
>* Disable ODT: 0.11A
> Qualitative results: The Orangecrab is noticeably cooler now and the memory tests still pass. Worth doing if you're concerned about heat

- [ ] Verify default ODT configuration (should default to Point to Point), disabling it should be possible with the MR registers (disabling RTT_NOM and enabling RTT_WR).
- [ ] Verify DQ  termination in Platform file.

enjoy-digital, 2020-08-28:
Current reduced to 0.12A @ 48MHz and 0.17A @96MHz with https://github.com/litex-hub/litex-boards/commit/beccecf59f313a674a18bacab0b24e6ecd6e2bb7.

closed 2020-08-28 by enjoy-digital

gregdavill, 2020-08-30:
Thanks for doing some tests!

I see you've manually driving every pin in `vccio`
```
self.comb += ddram_pads.vccio.eq(0b111111)
```

I feel like a fool now :P. On my example SoC design for CP I'd simply done this:
```
self.comb += ddram_pads.vccio.eq(1)
```
It's now so obvious why this added so much extra power consumption! -_-
Since it was only driving bit 0, The other VCCIO pins were driven LOW

emeb, 2020-08-30:
I remember seeing that  ddram_pads.vccio.eq(1) construct when I was testing current draw. It seemed odd to me, but since I was just disabling it I didn't chase it down further. Makes sense though.
Amazing/lucky that shorting a bunch of low outputs to VCCIO didn't do more than warm things up a bit.

emeb, 2020-08-30:
Interesting data point to add to the original post in this thread: I restored the virtual VCCIO hookups with the full 6-bits of '1' and re-ran the current draw test. Strangely enough it reduced the current even further:

* Enable Virtual VCCIO with all six bits high : 0.08A

I don't have a good explanation of why the current dropped with this change. I did confirm that prior to this change the current draw was consistent with earlier tests.