enjoy-digital/litedram #219: Core Question

vasimr, 2020-10-02:
Hi,

Can the DRAM controller core be generated without a CPU inside? 
I would like to have an external CPU that performs the initialization and calibration.

Thanks in advance.

enjoy-digital, 2020-10-02:
Hi @vasimr,

yes it's possible to do so, if you are looking for similar example, you can look https://github.com/antonblanchard/microwatt/tree/master/litedram/gen-src. Microwatt is integrating LiteDRAM in a similar way: the core is generated with` litedram_gen` without a CPU (`"cpu":   "None"`) and Microwatt is then doing the initialization through the Wishbone interface that is exposed.

vasimr, 2020-10-02:
Perfect, thanks! That worked perfectly!
One question regarding using an external interface. Does the time between writes matter for initialization and calibration? Or can it be done at the leisure of the external CPU? 

I have a few additional quick questions if you don't mind (I can email them instead or create another issue if you prefer).
1) Is there a way to set the RAM module parameters via the WB bus instead of having them pre-defined? (could be helpful for plug+play DIMM compatibility)
2) Are there any additional clocking / PLL / MMCM constraints that need to be considered when using a stand alone controller?
3) Is there a way to generate two controllers (for a dual channel configuration) and have them share the same interface clock? (generating two verilog files is fine, but I would like their native / wishbone / AXI to have the same clock to avoid additional clock bridging)
4) Are 64/128-bit AXI / Wishbone buses supported?
5) What's the performance of the multi-port bridge? Does it handle bursting (e.g. for AXI)? Will it have a substantial performance hit or would it be better to build one myself off of the native interface?

enjoy-digital, 2020-10-02:
The only requirement for the initialization/calibration is to do it before using the memory :)

For the others questions:
1) Not yet, we have plans for that, but for now the parameters are not dynamic but defined at build time.
2) You need to be sure to constraint the input clock correctly. The toolchain will then be able to propagate/derivate the other constraints. For the electrical constraints, please look at the one defined in` litex-boards` platforms for similar boards/configurations: https://github.com/litex-hub/litex-boards/tree/master/litex_boards/platforms
3)  yes it should be fine, it not please report it and we'll try to fix that together. (i'm also planning to test similar configurations in the next weeks).
4) For AXI it's only supporting the native controller's data width for now (there are plans to support data-width adaptation) but for Wishbone this should already be possible. (eventually using LiteX's WishboneConverter https://github.com/enjoy-digital/litex/blob/master/litex/soc/interconnect/wishbone.py#L303).
5) The arbiter is currently simple (RoundRobin on each banks) but should be convenient for most use cases. We have plans to improve it with https://github.com/enjoy-digital/litedram/issues/209 that should be implemented in the next weeks. So maybe you could start with it and get back to individuals ports if performance is not satisfying for your use case.

vasimr, 2020-10-02:
Thanks for the answers!

2. I'm not sure what I should be looking at there. They mostly look like pin IO constraints. Is that all that's needed for the tool to infer the correct settings? 
3. What would be the best way to go about implementing a shared clock? Should the PLL be removed from one of the modules and then fed the same signals from the other module? Would that be best done with only the user_clock (I believe it's connected to the 1st buffer) and then still instantiate two PLLs? should the clock be taken before or after the buffer? (before could cause the two clocks to go out of sync, after could put too much load on a the clock buffer)
4./5. I didn't notice that, but now I see that they are all the native width, so I will probably have to create a bridge / arbiter myself, which is fine. Thinking about it, I need something more niche than round-robin. I would start with the default one, however, for an x32 device, a 256-bit AXI / wishbone bus is a bit too wide
6. Can the IO to sysclk ratio be changed? If I understand correctly, it's 4:1 by default, however, I think 2:1 would provide better results for my application (allows for a higher bus frequency while dropping the DRAM IO frequency)

enjoy-digital, 2026-11-09:
Hi @vasimr,

sorry for the very late follow-up. Here are answers to the remaining questions, with the current state of the code:

2. Yes, the only timing constraint needed is a `create_clock` on the core's `clk` input at `input_clk_freq`: the generated XDC doesn't define it, and Vivado then derives the MMCM output clocks. The rest is IO constraints for the DRAM pins (IO standards, pin locations, VREF/termination), like in the `litex-boards` platforms. If you integrate the Verilog in your own project, also add the generated `litedram_core.xdc` (it holds the false paths of the reset synchronizers). This is now documented in #476.

3. With two cores generated separately, each has its own CRG/MMCM and its `user_clk` output: the two are at the same frequency but not in phase, so they have to be treated as two clock domains. For a single interface clock, the simplest is to build the two controllers in one LiteX design (one CRG, two PHYs and two `LiteDRAMCore`) instead of two `litedram_gen` cores. Otherwise, use one core's `user_clk` as your interface clock and cross the other core's port with an async FIFO.

4. Ports now support other data widths: `data_width` can be set on `native`, `wishbone`, `avalon` and `axi` ports in `litedram_gen`, and the crossbar converts to the controller's data width. So a 32/64/128-bit AXI or Wishbone port is fine on an x32 device.

5. Arbitration is still round-robin per bank. Improving it is #209, with a prototype in #474 (not merged yet). For a custom policy, you can still build your own arbiter on a single native port: see the [native port protocol](https://github.com/enjoy-digital/litedram/blob/master/doc/gen.md#native-port-protocol).

6. The ratio depends on the PHY and memory type. On 7-Series, DDR3 is 1:4 only (1:2 is for DDR2). On ECP5 and Gowin GW2A it's 1:2, and GW5A can do 1:2 or 1:4 (`sdram_rate`).

For 1: module parameters are still defined at build time.

Since the original question is answered, I'm closing this. Please open a new issue for any follow-up.

closed 2026-11-09 by enjoy-digital