enjoy-digital/litedram #220: Native Interface Documentation and Simulation

vasimr, 2020-10-03:
Hi,

Does there exist any documentation on how the native interface sequencing works?
It looks similar to the native interface that the Xilinx MIG uses, however, looking at the schematic generated, it appears that there are no intermediate registers so it's not clear how long the signals must / will remain valid. 

Also, is there a way to generate a simulation model for the interface, rather than doing a full system simulation?

Thanks in advance!

enjoy-digital, 2020-11-09:
Hi @vasimr,

i'll try to add documentation for it in the next days.

tommythorn, 2020-11-18:
I was looking for this as well.  I was trying to see if there were any support for split transactions, and if, how many could be outstanding.  While I don't read migen, it doesn't seems like there is.

enjoy-digital, 2026-11-06:
Hi @vasimr, @tommythorn,

sorry for the very late answer. This is now documented in [doc/gen.md, "Native port protocol"](https://github.com/enjoy-digital/litedram/blob/master/doc/gen.md#native-port-protocol). In short:
- **Command:** hold `cmd_valid`, `cmd_we` and `cmd_addr` until `cmd_ready`. `cmd_ready` is combinational from `cmd_valid`, so don't derive `cmd_valid` from it.
- **Write data:** `wdata_valid` is ignored. Present the data from the command acceptance until the single-cycle `wdata_ready` pulse, when it is sampled. A FIFO popped by `wdata_ready` is the usual implementation.
- **Read data:** `rdata_data` is only valid during the single-cycle `rdata_valid` pulse, and `rdata_ready` is ignored, so capture or buffer it on that cycle.
- **Split transactions:** several commands can be outstanding, up to `cmd_buffer_depth` per bank. They complete in order per port, and there are no transaction IDs.

For simulation, `litedram_gen --sim` gives the same core around a DFI-level DRAM model, to instantiate in your own testbench. With `block_until_ready` (the default), the ports stay blocked until the `ddrctrl_init_done` register is written; see ["Initialization with `cpu: None`"](https://github.com/enjoy-digital/litedram/blob/master/doc/gen.md#initialization-with-cpu-none).

Closing, feel free to re-open if something is missing.

closed 2026-11-06 by enjoy-digital