enjoy-digital/litedram #225: Wrong DQS pattern preamble/postamble

jedrzejboczar, 2020-12-10:
While working on https://github.com/enjoy-digital/litedram/pull/224 I'm using S7DDRPHY as the main reference, and I noticed that it is missing proper preamble/postamble https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/s7ddrphy.py#L214-L215. Currently it sends a full sysclk cycle of DQS pattern. But DRAM seems to be working correctly anyway (tested on Arty).

Another problem is that, at least from what I understand, that preamble/postamble are reversed here: https://github.com/enjoy-digital/litedram/blob/master/litedram/common.py#L164-L169. As LSB is transmitted first, it should probably be `preamble=0b01010100`, `postamble=0b00010101`.

I am adding the preamble/postamble generation to LPDDR4PHY and it seems that bigger latency is needed to propagate the change in DQS pattern. When I find a good solution I will create a PR, for now I just wanted to write an issue about it.

enjoy-digital, 2020-12-10:
Thanks @jedrzejboczar. This indeed needs to be investigated since I had to disable it while doing other improvements (hence the FIXME). The order should fine: in the current code, preamble is asserted the cycle **before** the transfer, so a pattern of `0b00010101` will set DQS to `0` the cycles just before the effective transfer. Same for postamble that is asserted the cycle **after** the transfer. The current issue is probably related to the latency, but a review of the code and simulations as you are doing will be very useful.

jedrzejboczar, 2020-12-10:
Ah, ok, I think I understand it now, thanks for clearing this up.

enjoy-digital, 2026-11-06:
Status update: the pattern order is fine, as discussed above. The missing preamble/postamble is still current on `S7DDRPHY` and `USDDRPHY`: the output enable covers these cycles, but `DQSPattern` keeps toggling (the `# FIXME` lines). Writes work on the boards using these PHYs, so this is low priority, but it is not fully clean. Enabling it needs the DQS pattern aligned with the output enable and then a test on hardware, so I'm keeping this open.