enjoy-digital/litedram #233: Add support for DDR3 in MAX10

rdolbeau, 2021-02-15:
Currently the MAX10 FPGA is supported on some boards (e.g. DE10 Lite), but they all use SDR memory & the GENSDRPHY in litedram.

Some cheap yet full-featured boards like the [Arrow Deca ($37 as of now)](https://www.arrow.com/en/products/deca/arrow-development-tools) use DDR3 memory (512 MiB) and could be excellent (Linux-on)-LiteX platform.

Would it be possible to implement a DDR3 PHY for the MAX10 in litedram, like their is for other FPGA (A7DDRPHY for the Artix A7, ECP5DDRPHY for the ECP 5, ...) so the such boards become usable with LiteX?

enjoy-digital, 2021-02-16:
Hi @rdolbeau,

that would indeed be nice to improve support for Intel FPGAs. Developing a new PHY still requires some work and the others PHYs have generally been developed when working on specific systems for clients. So I'm sure there will be an occasion to develop this, but can't guarantee the timeline :)

rdolbeau, 2021-02-16:
@enjoy-digital Yes I thought as much. Too bad, they seem to be nice cheap boards, but without LiteX what's the point ? :-)

Intel has a tool to generate a controller (UniPHY), the controller uses an Avalon bus if I read the documentation correctly - and it seems to me LiteX support Avalon. Could that be an easier way to support the board (and similar one), theoretically speaking ?

enjoy-digital, 2021-02-16:
@rdolbeau: This is indeed a possible/hybrid solution. I remember doing this in the past on some Cyclone devices. This will work fine but will not use LiteDRAM. LiteX has a minimal Avalon-ST support (with similar aims than AXI-ST), but not Avalon-MM that will be required here. It should however be close to Wishbone/AXI-MMAP so this could be a way to add a WIshbone-->Avalon-MM bridge to LiteX :) I can order a Arrow Deca if you want to work together on this (was also planning to get one).