enjoy-digital/litedram #235: simple example to read-write for the standalone core?

particlerain, 2021-03-09:
Hi, thanks for the awesome Litex!
I would like to use the standalone litedram core in my design but need a simple example to read-write in verilog. Can you advice something?

enjoy-digital, 2021-03-11:
Hi @particlerain,

thanks for the feedback. I would recommend using the standardized interfaces to start using the core: Wishbone or AXI. Once `init_done` signals is set (and with `init_error` not set), you can start using the core as you would do with any other DRAM controller. 

You can find an example of integration in these projects:
- https://github.com/chipsalliance/Cores-SweRVolf/tree/master/cores/litedram
- https://github.com/antonblanchard/microwatt/tree/master/litedram

particlerain, 2021-03-13:
Exactly what i was looking for! Thank you.

Btw, in the gen.py you mentioned: 

`Current version of the generator is limited to:
- DDR3 on Lattice ECP5 FPGAs.
- DDR2/DDR3 on Xilinx 7-Series FPGAs.
- DDR4 on Xilinx Ultascale(+) FPGAs.`

Are you hopefully planning to update the gen.py for the standalone core to also include the sdrams? For example IS42S16160?

enjoy-digital, 2021-03-13:
Yes this could be done, the current version of the generator only supports the cases we had a need for.

particlerain, 2021-03-13:
> 
> 
> Yes this could be done, the current version of the generator only supports the cases we had a need for.
Great! 

Thanks for your replies. I am closing this now.

closed 2021-03-13 by particlerain