enjoy-digital/litedram #246: Race condition when prematurely aborting Wishbone transactions

jfng, 2021-04-26:
Context: simulating a LiteDRAM core (generated by `litedram.gen`) with a Wishbone user interface.

When aborting a transaction (e.g. a read at `0x17018`), then following it with another transaction (e.g. a read at `0x17020`), the latter may return invalid data.

Here is a [ZIP archive](https://github.com/enjoy-digital/litedram/files/6379404/repro.zip) with a reproduction of this bug (using nMigen+Yosys CXXRTL). To run it:
```
make; ./repro_driver
gtkwave dump.gtkw
```

AFAIU, the aborted transaction caused side effects that were not completely resolved once the next one was initiated.

enjoy-digital, 2021-04-27:
Thanks @jfng for reporting and the repro. The code was not supporting Wishbone aborts but I should have added support for it while doing some simplifications in https://github.com/enjoy-digital/litedram/commit/bd80053ebf078a65f7c4614fb110fc1905166d4e. I don't have CXXRTL installed currently, could you re-run the test on your side? I'll install CXXRTL if you still reproduce the issue.

jfng, 2021-04-28:
Thank you ! The fix has worked for me with these two modifications:
```diff
@@ -56,11 +56,12 @@ class LiteDRAMWishbone2Native(Module):
         )
         self.comb += [
             port.wdata.valid.eq(wishbone.stb & wishbone.we),
-            If(ratio < 1, If(~fsm.ongoing("WRITE"), port.wdata.valid.eq(0))),
+            If(~fsm.ongoing("WRITE"), port.wdata.valid.eq(0)),
             port.wdata.data.eq(wishbone.dat_w),
             port.wdata.we.eq(wishbone.sel),
         ]
         fsm.act("WRITE",
             NextValue(aborted, ~wishbone.cyc | aborted),
             If(port.wdata.valid & port.wdata.ready,
                 wishbone.ack.eq(wishbone.cyc & ~aborted),
@@ -71,7 +72,7 @@ class LiteDRAMWishbone2Native(Module):
         fsm.act("READ",
             NextValue(aborted, aborted | ~wishbone.cyc),
             If(port.rdata.valid,
-                wishbone.ack.eq(~aborted | wishbone.cyc),
+                wishbone.ack.eq(wishbone.cyc & ~aborted),
                 wishbone.dat_r.eq(port.rdata.data),
                 NextState("CMD")
             )
```
(though I'm not sure about the correctness of my `ratio` change)

jfng, 2021-04-28:
Also, if you wish to run the repro yourself, you'll need to update my Makefile with the following:
(due to a CXXRTL bug)
```diff
10c10
<       yosys repro.il build/gateware/litedram_core.v -o repro.cc
---
>       yosys repro.il build/gateware/litedram_core.v -p "write_cxxrtl -O4 repro.cc"
```

enjoy-digital, 2021-04-28:
Thanks for the feedback, sorry for the typos, this should be fixed in https://github.com/enjoy-digital/litedram/commit/6755c8438f8233b03aecf2f0bcd6081c24a755d6. Would you mind just doing the test with upstream and confirm it's OK? Thanks.

jfng, 2021-04-28:
The upstream works for me, thanks !

enjoy-digital, 2021-04-28:
Thanks for the feedback. I'm closing then but will probably play later with your repro to explore a bit more CXXRTL.

closed 2021-04-28 by enjoy-digital