enjoy-digital/litedram #255: Fix test_lpddr4.VerilatorLPDDR4Tests.test_lpddr4_sim_x2rate_no_cache.

enjoy-digital, 2021-06-08:
Recent LiteX/LiteDRAM changes probably broke this test, it would need to be updated. The test has been disabled for now with https://github.com/enjoy-digital/litedram/commit/377d6fac6c2fd5718061d932f3a838087a159ee8. @jedrzejboczar could  you eventually have a look at this? Thanks.

jedrzejboczar, 2021-06-08:
From initial test it seems that there is an issue in the commands send during the init sequence. I will take a closer look to see where the problem is.

jedrzejboczar, 2021-06-09:
After more tests I found that the issue is that `cdelay` during SDRAM init sequence results in too short delay, violating tINIT3 timing (defined [here](https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/lpddr4/sim.py#L198)). This delay is defined in [init.py for LPDDR4](https://github.com/enjoy-digital/litedram/blob/master/litedram/init.py#L622). The generated sequence in `sdram_phy.h` looks like this:
```c
static inline void init_sequence(void)
{
	/* Assert reset */
	sdram_dfii_pi0_address_write(0x0);
	sdram_dfii_pi0_baddress_write(0);
	sdram_dfii_control_write(DFII_CONTROL_ODT);
	cdelay(20);

	/* Release reset */
	sdram_dfii_pi0_address_write(0x0);
	sdram_dfii_pi0_baddress_write(0);
	sdram_dfii_control_write(DFII_CONTROL_ODT|DFII_CONTROL_RESET_N);
	cdelay(400000);

	/* Bring CKE high */
	sdram_dfii_pi0_address_write(0x0);
	sdram_dfii_pi0_baddress_write(0);
	sdram_dfii_control_write(DFII_CONTROL_CKE|DFII_CONTROL_ODT|DFII_CONTROL_RESET_N);
	cdelay(400);
```
In the simulation output it can be seen that the delays are wrong. The difference defined by `cdelay(20)` is the time between `[INFO] RESET asserted` and `[INFO] RESET released`:
```
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
[      2000052500 ps] [INFO] FSM: RESET -> EXIT-PD
[      2002055000 ps] [INFO] FSM: EXIT-PD -> MRW
[      2363430000 ps] [INFO] RESET asserted
[      2363430000 ps] [INFO] CKE falling edge
[      2363530000 ps] [INFO] FSM reset
[      2363532500 ps] [INFO] FSM: MRW -> RESET
[    130366410000 ps] [INFO] RESET released
[    130497310000 ps] [INFO] CKE rising edge
[    130497310000 ps] [WARN] tINIT3 violated: CKE set HIGH too fast after RESET being released
Read leveling:
  m0, b00: |0| delays: -
  m0, b01: |[    132366412500 ps] [INFO] FSM: RESET -> EXIT-PD
[    132368415000 ps] [INFO] FSM: EXIT-PD -> MRW
[    132390030000 ps] [WARN] Only MRW/MRR commands expected before ZQ calibration
[    132390030000 ps] [WARN] MRW=0 REF=0 ACT=0 PRE=0 CAS=1 MPC=0
```
From the logs this time is ~128 ms, which is clearly too long for `cdelay(20)`. The second time, defined by `cdelay(400000)` is between `[INFO] RESET released` and `[INFO] CKE rising edge`, and it is ~130 us which is too low (should be ~2 ms).

I was testing what could cause the problem, and finally I changed [VexRiscV CPU variant](https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/lpddr4/simsoc.py#L127) from "minimal" to "lite", which makes the timings correct, the same for "standard" variant. I wonder what causes the issue. I checked the output of `riscv64-elf-objdump -D build/sim/software/bios/bios.elf`, but the code for `cdelay` and the init sequence in `sdram_init` looks the same.

For reference here is the corresponding simulation output for `cpu_variant="lite"`:
```
[           50000 ps] [INFO] RESET released
[           50000 ps] [WARN] tINIT1 violated: RESET deasserted too fast
[           50000 ps] [INFO] CKE rising edge
[           50000 ps] [WARN] tINIT3 violated: CKE set HIGH too fast after RESET being released
[          100000 ps] [INFO] FSM reset
--========== Initialization ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
[      1317410000 ps] [INFO] RESET asserted
[      1317410000 ps] [INFO] CKE falling edge
[      1317510000 ps] [INFO] FSM reset
[      1321430000 ps] [INFO] RESET released
[      3321432500 ps] [INFO] FSM: RESET -> EXIT-PD
[      3323435000 ps] [INFO] FSM: EXIT-PD -> MRW
[     65322190000 ps] [INFO] CKE rising edge
[     65387060000 ps] [INFO] MRW: MR[ 1] = 0x14
[     65419940000 ps] [INFO] MRW: MR[ 2] = 0x09
[     65452820000 ps] [INFO] MRW: MR[ 3] = 0x11
[     65485700000 ps] [INFO] MRW: MR[11] = 0x22
[     65518540000 ps] [INFO] MRW: MR[12] = 0x55
[     65551300000 ps] [INFO] MRW: MR[13] = 0x00
[     65584140000 ps] [INFO] MRW: MR[14] = 0x55
[     65616880000 ps] [INFO] MPC: ZQC_START
[     65616882500 ps] [INFO] FSM: MRW -> ZQC
[     65649360000 ps] [INFO] MPC: ZQC_LATCH
[     65649362500 ps] [INFO] FSM: ZQC -> NORMAL
Read leveling:
  m0, b00: |[     65799480000 ps] [INFO] ACT: bank=0 row=     0
[     65816995000 ps] [INFO] MASKED-WRITE: bank=0 row=     0 col=   0
```

So a fix for this test would be to change `cpu_variant` to "lite" here: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/lpddr4/simsoc.py#L127. But it might be worth understanding why variant "minimal" causes such incorrect delays.

enjoy-digital, 2021-06-09:
@jedrzejboczar: Thanks for looking at this and the explanation. If the issue is related to VexRiscv Minimal variant, we could indeed maybe workaround it it in LiteDRAM and open another issue in LiteX specific to this. Would you mind providing a minimal repro demonstrating the issue with VexRiscv Minimal and creating an issue in LiteX?

enjoy-digital, 2021-07-02:
The variant has been changed to `lite` with https://github.com/enjoy-digital/litedram/commit/a3aa4907f11f654dc2df58e13903ec99fad69b6f. We'll continue the investigation with the `minimal` variant in https://github.com/enjoy-digital/litex/issues/937.

closed 2021-07-02 by enjoy-digital