enjoy-digital/litedram #260: SDRAM support for standalone core

ombhilare999, 2021-06-26:
I see for standalone generation there is this file: [gen.py](https://github.com/enjoy-digital/litedram/blob/master/litedram/gen.py#L532)
But it has DDR3,  DDR2/DDR3 and DDR4 support. Will the SDRAM standalone support will be a complicated job?
Can you direct me to how to add standalone SDRAM support?
```
Current version of the generator is limited to:
- DDR3 on Lattice ECP5 FPGAs.
- DDR2/DDR3 on Xilinx 7-Series FPGAs.
- DDR4 on Xilinx Ultascale(+) FPGAs.
```

enjoy-digital, 2021-06-29:
Hello @ombhilare999,

we indeed only implemented in the generator the use cases we had a need for. I'll try to provide a skeleton for SDRAM integration and/or directions in the next days. Supporting it should not be complicated.

ombhilare999, 2021-07-01:
Hello @enjoy-digital ,
Thank you, The skeleton will definitely help. I will be looking forward to this thread.

enjoy-digital, 2021-07-02:
Initial SDRAM support has been added with https://github.com/enjoy-digital/litedram/commit/317072a1982d8f20166be92874d265ac603e46b9 and https://github.com/enjoy-digital/litedram/commit/a11d1b870d23a9647e871afa116777cede3d532d and a configuration example has been added for the ULX3S. Adapting the example for another board should be easy.

With SDRAM, the clocking has to be handled externally and looking at LiteX-Boards targets can be useful for this.

closed 2021-07-02 by enjoy-digital