enjoy-digital/litedram #261: SDRAMPHYModel should pick sane default PHY settings

thirtythreeforty, 2021-07-02:
First, thank you for this awesome library!

I have a goal of simulating my entire SoC in Verilator, with ECP5-specific PHYs swapped out for loopbacks or whatever.

Currently SDRAMPHYModel is difficult to use for this; I have to specify `settings`, which, based on [generous cribbing from litex_sim](https://github.com/enjoy-digital/litex/blob/master/litex/tools/litex_sim.py#L95), seems to only be based on the memtype of the provided module.

If I specify a module, it should be possible for the model to infer these. As a DRAM noob who only knows how to lay it out on a board, this would really simplify dropping litedram into my SoC model.

Preferred instantiation would be something like:

```
            sdram_clk_freq = clk_freq
            sdram_module = MT41K256M16(sdram_clk_freq, '1:2')
            sdram_verbosity = 0

            self.submodules.sdrphy = SDRAMPHYModel(
                module    = sdram_module,
                clk_freq  = sdram_clk_freq)

            self.add_sdram("sdram",
                phy                     = self.sdrphy,
                module                  = sdram_module,
                origin                  = self.mem_map["sdram"],
                size                    = kwargs.get("max_sdram_size", 0x40000000),
                **l2_args,
            )
```

enjoy-digital, 2021-07-08:
Thanks @thirtythreeforty for the feedback/suggestion. It indeed makes sense. When working on allowing the simulation of the various SDRAM modules last year we focused on getting it working, but that's now true that we should now simplify the integration in simulation. I'll try to look at this very soon.

enjoy-digital, 2021-07-08:
This has been implemented with:
- https://github.com/enjoy-digital/litedram/commit/daf2cb7d396e711da5e172b856be253e8f0eb139
- https://github.com/enjoy-digital/litex/commit/fced79631ed3f18baaa0e03c86bc4793012b8b70
- https://github.com/enjoy-digital/litedram/commit/894c7fb49e294aa94a4f33b1bbb71cd411124f2b
- https://github.com/enjoy-digital/litex/commit/9a51dfb50e878c1da4342ef681e9f228b18d1d0c

The only difference with your suggestion is that SDRAMPHYModel still needs to know the total SDRAM data_width (which is not provided by the SDRAM module), so the minimal `SDRAMPHYModel`iinstance is:

```python3
self.submodules.sdrphy = SDRAMPHYModel(
    module       = sdram_module,
     data_width  = sdram_data_width,
     clk_freq    = sdram_clk_freq)
```

thirtythreeforty, 2021-07-08:
Excellent! That makes sense. Much appreciated

thirtythreeforty, 2021-07-09:
Unfortunately I get an error:

```
  File "/home/georgev/Code/sonata-gateware/sonata_gateware/sonata_soc.py", line 146, in __init__
    self.submodules.sdrphy = SDRAMPHYModel(
  File "/home/georgev/Code/sonata-gateware/.venv/lib/python3.9/site-packages/litedram/phy/model.py", line 513, in __init__
    settings = get_sdram_phy_settings(
  File "/home/georgev/Code/sonata-gateware/.venv/lib/python3.9/site-packages/litedram/phy/model.py", line 412, in get_sdram_phy_settings
    cl, cwl         = get_default_cl_cwl(memtype, tck)
NameError: name 'get_default_cl_cwl' is not defined
```

when trying to instantiate it:
```py
            self.submodules.sdrphy = SDRAMPHYModel(
                module     = sdram_module,
                data_width = 16,
                clk_freq   = sdram_clk_freq)
```

enjoy-digital, 2021-07-09:
Thanks for the feedback, this should be fixed with https://github.com/enjoy-digital/litedram/commit/fd8d6f8334df7fd5f38f4074b5311560ee9dc081.

thirtythreeforty, 2021-07-19:
Tested and working! Thanks!

closed 2021-07-19 by thirtythreeforty