enjoy-digital/litedram #265: Figure out how to hook LiteDRAM to the WDDR PHY

mithro, 2021-07-21:
@waviousllc recently released an [DDR PHY under an Apache 2.0 license](https://github.com/waviousllc/wav-lpddr-hw). The PHY is currently being used with proprietary memory controllers and it would be awesome to see it work with LiteDRAM instead.

As this is the first time LiteDRAM will be targeting a ASIC PHY rather an FPGA, it is likely there will be a bunch of experimentation needed to make things work, hence I'm creating this tracking issue.

mithro, 2021-07-21:
See also https://github.com/waviousllc/wav-lpddr-hw/issues/12

enjoy-digital, 2026-11-06:
Closing as not planned: I have no plan or hardware for this, the WDDR repository has had no pushes since July 2021, and there has been no activity here. Happy to look at a PR if someone takes it on.

closed 2026-11-06 by enjoy-digital