enjoy-digital/litedram #281: Help generating DDR3 Verilog module for Digilent NexysVideo Artix-7 FPGA

tambewilliam, 2021-10-10:
I have had no luck generating a working DDR3 Verilog module for Digilent NexysVideo Artix-7 FPGA using:
```
{
    # General ------------------------------------------------------------------
    "speedgrade": -1,      # FPGA speedgrade
    "cpu":        "None",  # CPU type (ex vexriscv, serv, None)
    "memtype":    "DDR3",  # DRAM type

    # PHY ----------------------------------------------------------------------
    "cmd_latency":     0,             # Command additional latency
    "sdram_module":    "MT41K256M16", # SDRAM modules of the board or SO-DIMM
    "sdram_module_nb": 2,             # Number of byte groups
    "sdram_rank_nb":   1,             # Number of ranks
    "sdram_phy":       "A7DDRPHY",    # Type of FPGA PHY

    # Electrical ---------------------------------------------------------------
    "rtt_nom": "60ohm", # Nominal termination
    "rtt_wr":  "60ohm", # Write termination
    "ron":     "34ohm", # Output driver impedance

    # Frequency ----------------------------------------------------------------
    "input_clk_freq":   100e6, # Input clock frequency
    "sys_clk_freq":     100e6, # System clock frequency (DDR_clk = 4 x sys_clk)
    "iodelay_clk_freq": 200e6, # IODELAYs reference clock frequency

    # Core ---------------------------------------------------------------------
    "cmd_buffer_depth": 16, # Depth of the command buffer

    # User Ports ---------------------------------------------------------------
    "user_ports": {
        "wishbone_0" : {
            "type": "wishbone",
        },
    },
}
```

The generated module gets successfully initialized through its wb_ctrl_* interface, but transactions through its user_port_wishbone_* interface never get acknowledged.

Find attached  in [litedram_nexysvideo.tar.gz](https://github.com/enjoy-digital/litedram/files/7316693/litedram_nexysvideo.tar.gz) following generated files using `litedram/gen.py`:
```
build
 |-gateware
 | |-litedram_core.xdc
 | |-litedram_core.v
 | |-litedram_core.tcl
 |-software
 | |-include
 | | |-generated
 | | | |-sdram_phy.h
 | | | |-soc.h
 | | | |-git.h
 | | | |-csr.h
 | | | |-mem.h
 ```

Could anyone verify that a module that uses wishbone and generated for Digilent NexysVideo Artix-7 FPGA using `litedram/gen.py` works ?

tambewilliam, 2021-10-29:
@enjoy-digital , I was able to generate a working verilog module that uses wishbone after following revert:
git checkout bd80053ebf078a65f7c4614fb110fc1905166d4e~ litedram/frontend/wishbone.py

enjoy-digital, 2021-11-01:
@tambewilliam: Thanks for the feedback. I reopen this since will look at it.

shingarov, 2022-09-18:
Has this ever been fixed?  According to https://github.com/antonblanchard/microwatt/issues/363 the current master works on NexysVideo.  Perhaps only the issue needs closing?  I verified today and memtest fails on Genesys2 for me.

enjoy-digital, 2026-11-06:
Sorry for the very late answer. I generated your exact YAML on current master with `--sim` and ran it in Verilator: 256 writes and 256 reads through the Wishbone port were all acked with correct data. That uses the simulation PHY, so it does not cover the real board. One thing can give your symptom: with `cpu: None` and `block_until_ready` (the default), the port never acks until software writes the `ddrctrl_init_done` CSR, and `init_sequence()` does not do this (now documented in #469). `wishbone.py` has also changed since your report, including a fix for a write hang when `stb` drops after the command is accepted (19f9e28). Closing, please re-open if it still fails on master.

closed 2026-11-06 by enjoy-digital