enjoy-digital/litedram #285: DDR3 Memory on K7325 based board randomly fails

tongchen126, 2021-11-11:
I have ported litex-board to a K7325T based board. I have checked many times that the pinout is correct. But the DDR3 ram randomly fails. Sometimes when it passes the boot memory check, then it goes smoothly. But sometimes, it will just fail the pre-boot memory test. I got no idea why sometimes it works and works fine all way down, and sometimes it will just crash.
[image]
[image]
[image]

tongchen126, 2021-11-11:
#!/usr/bin/env python3

#
# This file is part of LiteX-Boards.
#
# Copyright (c) 2020 Mark Standke <mstandke@cern.ch>
# SPDX-License-Identifier: BSD-2-Clause

import os
import argparse

from migen import *

from litex_boards.platforms import k7325t

from litex.soc.cores.clock import *
from litex.soc.integration.soc_core import *
from litex.soc.integration.builder import *
from litex.soc.cores.led import LedChaser
from litepcie.software import generate_litepcie_software

from litedram.phy import s7ddrphy

# CRG ----------------------------------------------------------------------------------------------

class _CRG(Module):
    def __init__(self, platform, sys_clk_freq):
        #self.rst = Signal()
        self.clock_domains.cd_sys    = ClockDomain()
        self.clock_domains.cd_sys4x  = ClockDomain(reset_less=True)
        self.clock_domains.cd_idelay = ClockDomain()

        # # #

        self.submodules.pll = pll = S7MMCM(speedgrade=-2)
        self.comb += pll.reset.eq(~platform.request("cpu_reset_n"))# | self.rst)
        pll.register_clkin(platform.request("clk200"), 200e6)
        pll.create_clkout(self.cd_sys,    sys_clk_freq)
        pll.create_clkout(self.cd_sys4x,  4*sys_clk_freq)
        pll.create_clkout(self.cd_idelay, 200e6)
        platform.add_false_path_constraints(self.cd_sys.clk, pll.clkin) # Ignore sys_clk to pll.clkin path created by SoC's rst.

        self.submodules.idelayctrl = S7IDELAYCTRL(self.cd_idelay)

# BaseSoC ------------------------------------------------------------------------------------------

class BaseSoC(SoCCore):
    def __init__(self, sys_clk_freq=int(125e6), with_ethernet=False, with_led_chaser=False,
                 with_pcie=False, with_sata=False, **kwargs):
        platform = k7325t.Platform()

        # SoCCore ----------------------------------------------------------------------------------
        SoCCore.__init__(self, platform, sys_clk_freq,
            ident          = "LiteX SoC on KX2",
            ident_version  = True,
            **kwargs)

        # CRG --------------------------------------------------------------------------------------
        self.submodules.crg = _CRG(platform, sys_clk_freq)

        # DDR3 SDRAM -------------------------------------------------------------------------------
        if not self.integrated_main_ram_size:
            from litedram.modules import MT8JTF12864,MT41J256M8,MT16KTF1G64HZ
            self.submodules.ddrphy = s7ddrphy.K7DDRPHY(platform.request("ddram_2R"),
                memtype      = "DDR3",
                nphases      = 4,
                sys_clk_freq = sys_clk_freq)
            self.add_sdram("sdram",
                phy           = self.ddrphy,
                module        = MT16KTF1G64HZ(sys_clk_freq, "1:4",speedgrade="1333"),
                l2_cache_size = kwargs.get("l2_size", 8192),
                size          = 0x40000000,
            )
        if with_pcie:
            from litepcie.phy.s7pciephy import S7PCIEPHY
            self.submodules.pcie_phy = S7PCIEPHY(platform, platform.request("pcie_x4"),
                data_width = 128,
                bar0_size  = 0x20000)
            self.add_pcie(phy=self.pcie_phy, ndmas=1)
        # Leds -------------------------------------------------------------------------------------
        if with_led_chaser:
            self.submodules.leds = LedChaser(
                pads         = platform.request_all("user_led"),
                sys_clk_freq = sys_clk_freq)
        for i in platform.request_all("user_led"):
            i.eq(1)
# Build --------------------------------------------------------------------------------------------

def main():
    parser = argparse.ArgumentParser(description="LiteX SoC on KX2")
    parser.add_argument("--build",        action="store_true", help="Build bitstream")
    parser.add_argument("--load",         action="store_true", help="Load bitstream")
    parser.add_argument("--sys-clk-freq", default=125e6,       help="System clock frequency (default: 125MHz)")
    parser.add_argument("--with-ethernet", action="store_true", help="Enable Ethernet support")
    parser.add_argument("--with-pcie",     action="store_true", help="Enable PCIe support")
    parser.add_argument("--driver",        action="store_true", help="Generate PCIe driver")
    parser.add_argument("--with-sata",     action="store_true", help="Enable SATA support (over SFP2SATA)")
    builder_args(parser)
    soc_core_args(parser)
    args = parser.parse_args()

    soc = BaseSoC(
        sys_clk_freq = int(float(args.sys_clk_freq)),
        with_ethernet=args.with_ethernet,
        with_pcie=args.with_pcie,
        with_sata=args.with_sata,
        **soc_core_argdict(args)
    )
    soc.add_spi_sdcard()

    builder = Builder(soc, **builder_argdict(args))
    builder.build(run=args.build)

    if args.driver:
        generate_litepcie_software(soc, os.path.join(builder.output_dir, "driver"))

    if args.load:
        prog = soc.platform.create_programmer()
        prog.load_bitstream(os.path.join(builder.gateware_dir, soc.build_name + ".bit"))

if __name__ == "__main__":
    main()

tongchen126, 2021-11-11:
I guess the problem might occur in the memory init stage. When it fails to pass the memtest, I tried sdram_init many times, and it would pass after numerous tries.

enjoy-digital, 2021-11-11:
Hi @tongchen126,

could you share the DDR3 calibration log? Could you also verify that the DCI Cascade or INTERNAL VREF constraints are correctly applied? If you have a MIG project, I would also recommend checking the constraints against it.

tongchen126, 2021-11-12:
Thanks for help. I have posted the link to the log, board files and mig constraints files. https://github.com/tongchen126/litedram_k7325t_log

tongchen126, 2021-11-12:
Is it possible to set the calibration value to fixed value, by modifying [litex/litex/soc/software/liblitedram/sdram.c](https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitedram/sdram.c).

gillham, 2022-02-02:
@tongchen126 have you made any progress on this?  I also ported LiteX to this board and didn't run across your work until I ran into the same issue and started searching.
I put my files here:  https://github.com/gillham/stlv7325 and also committed them to a fork of litex-boards.

If I can get this board to work reasonably, I was planning on doing a pull request from here: https://github.com/gillham/litex-boards/

I've tried to make a custom memory module based on SPD information published from Samsung for the module on my board.   This didn't appear to have any impact on the init/calibration.

The board I purchased is listed here:  https://www.aliexpress.com/item/1005001275162791.html

enjoy-digital, 2022-02-02:
@gillham, @tongchen126: If one of you could send me the hardware or give me remote access, I should probably be able to understand/fix this quickly.

gillham, 2022-02-03:
@enjoy-digital I'll email you about access to this board.

enjoy-digital, 2022-02-24:
@tongchen126: @gillham sent me a board and I have been able to look at the issue. During the read leveling, the transition from a failing pattern to a valid pattern does not seems as frank as on other Kintex7/Artix7 boards that have been tested in the past. To workaround this and allow a valid calibration, I added https://github.com/enjoy-digital/litex/commit/f6d6611a81bf9bc3015fefd280dad34fe455f2ed. Could you do a test on your board? The board support is now available in LiteX-Boards BTW.

tongchen126, 2022-02-25:
@enjoy-digital That is so great! I thought it was a hardware flaw before. I will give you feedback on the test.

tongchen126, 2022-02-25:
@enjoy-digital I have tested it 20 times, running at 100MHz, and it did not fail. And when it ran at 125MHz, it failed 4 times out of 10 tries. The write speed is around 168MiB/s and read speed 80 MiB/s at 100MHz.

enjoy-digital, 2022-02-25:
Thanks @tongchen126 for the feedback, I just reduced `sys_clk_freq` to 100MHz with https://github.com/litex-hub/litex-boards/commit/1623ba5942d7e6de6a81913229d2ee9a4982c51f. I never saw the pattern we are currently seeing on other boards and I really suspect this is related to this specific board, so that that we have a solution working at 100MHz with this board I suggest keeping this. Is it fine for you @tongchen126?

tongchen126, 2022-02-25:
@enjoy-digital Yes for sure. It should be the best freq currently for this board.

enjoy-digital, 2022-02-25:
@tongchen126: OK thanks. I'm closing this then. BTW I tested with another DDR3 memory have have the same symptoms, so this really seems related to the board.

closed 2022-02-25 by enjoy-digital

developandplay, 2022-04-26:
I just tried getting vexriscv and rocket running on the stlv7325 board.
I tried 50Mhz, 100Mhz and 150Mhz. However for all of them the memory test fails.
Are there any settings that I could tune like we did here https://github.com/enjoy-digital/litex/issues/933?

enjoy-digital, 2022-04-27:
@developandplay: Can you share the logs/build commands?

developandplay, 2022-04-27:
@enjoy-digital 
Sure here you go. Timing passed at all tested frequencies.
Interestingly for Rocket I consistently get 0 bus errors but for vexriscv I do get some.
Data errors consistently occur on both.

Vexriscv@ 50Mhz
`./litex/litex-boards/litex_boards/targets/aliexpress_stlv7325.py --build --load --cpu-type vexriscv_smp --sys-clk-freq 50e6 --with-sdcard --with-ethernet --csr-csv=csr.csv`
```
        __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 27 2022 09:08:19
 BIOS CRC passed (28fcddd8)

 LiteX git sha1: 3959ca39

--=============== SoC ==================--
CPU:		VexRiscv SMP-STANDARD @ 50MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 400MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 64
  Cmd/Clk scan (0-32)
  |00000000000000000000000000000000| best: 0
  Setting Cmd/Clk delay to 0 taps.
  Data scan:
  m0: |0000000000000000| delay: -
  m1: |0000000000000000| delay: -
  m2: |0000000000000000| delay: -
  m3: |0000000000000000| delay: -
  m4: |0000000000000000| delay: -
  m5: |0000000000000000| delay: -
  m6: |0000000000000000| delay: -
  m7: |0000000000000000| delay: -
Write latency calibration:
m0:0 m1:0 m2:6 m3:6 m4:0 m5:0 m6:6 m7:4 
Write DQ-DQS training:
m0: |11111111111100000000000000000| delays: 07+-07
m1: |11111111111100000000000000000| delays: 07+-07
m2: |000000000000000000000000000000000| delays: -
m3: |000000000000000000000000000000000| delays: -
m4: |11111111111100000000000000000| delays: 07+-07
m5: |11111111111100000000000000000| delays: 07+-07
m6: |000000000000000000000000000000000| delays: -
m7: |000000000000000000000000000000000| delays: -
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |11111111111100000000000000000000| delays: 05+-05
  m0, b02: |00000000000000001111111111111111| delays: 23+-08
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b02 delays: 24+-07
[... 480 lines cut ...]
  m6, b07: |00000000000000000000000000000000| delays: -
  best: m6, b02 delays: -
  m7, b00: |00000000000000000000000000000000| delays: -
  m7, b01: |00000000000000000000000000000000| delays: -
  m7, b02: |00000000000000000000000000000000| delays: -
  m7, b03: |00000000000000000000000000000000| delays: -
  m7, b04: |00000000000000000000000000000000| delays: -
  m7, b05: |00000000000000000000000000000000| delays: -
  m7, b06: |00000000000000000000000000000000| delays: -
  m7, b07: |00000000000000000000000000000000| delays: -
  best: m7, b02 delays: -
Switching SDRAM to hardware control.
Memtest at 0x80000000 (2.0MiB)...
  Write: 0x80000000-0x80200000 2.0MiB     
   Read: 0x80000000-0x80200000 2.0MiB     
  bus errors:  0/256
  addr errors: 0/8192
  data errors: 467497/524288
Memtest KO
Memory initialization failed

--============= Console ================--

litex> 
```

enjoy-digital, 2022-04-27:
@developandplay: I just did a test on my STLV325 board:

  ```
./aliexpress_stlv7325.py --uart-name=jtag_uart --cpu-type vexriscv_smp --sys-clk-freq 100e6 --with-sdcard --with-ethernet --build --load
litex_term jtag --jtag-config=openocd_xc7_ft232.cfg
  ```

Gives:
  ```
     __   _ __      _  __
       / /  (_) /____ | |/_/
      / /__/ / __/ -_)>  <
     /____/_/\__/\__/_/|_|
   Build your hardware, easily!

 (c) Copyright 2012-2022 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 27 2022 14:37:13
 BIOS CRC passed (3701457d)

 LiteX git sha1: 63cda6c7

--=============== SoC ==================--
CPU:		VexRiscv SMP-STANDARD @ 100MHz
BUS:		WISHBONE 32-bit @ 4GiB
CSR:		32-bit data
ROM:		128KiB
SRAM:		8KiB
L2:		8KiB
SDRAM:		1048576KiB 64-bit @ 800MT/s (CL-6 CWL-5)

--========== Initialization ============--
Ethernet init...
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 32
  Cmd/Clk scan (0-16)
  |0000000000000001| best: 14
  Setting Cmd/Clk delay to 14 taps.
  Data scan:
  m0: |111111110000000000000000| delay: -
  m1: |111111111100000000000000| delay: 00
  m2: |111111111110000000000000| delay: 00
  m3: |111111111111000000000000| delay: 00
  m4: |001111111111111111000000| delay: 02
  m5: |011111111111111110000000| delay: 01
  m6: |000111111111111111100000| delay: 03
  m7: |000111111111111111100000| delay: 03
Write latency calibration:
m0:6 m1:6 m2:6 m3:6 m4:6 m5:6 m6:6 m7:6 
Write DQ-DQS training:
m0: |11110000000000000000000000000| delays: 03+-03
m1: |11110000000000000000000000000| delays: 03+-03
m2: |11111000000000000000000000000| delays: 04+-04
m3: |11100000000000000000000000000| delays: 03+-03
m4: |11111100000000000000000000000| delays: 04+-04
m5: |11111000000000000000000000000| delays: 04+-04
m6: |11111100000000000000000000000| delays: 04+-04
m7: |11111110000000000000000000000| delays: 05+-05
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000000000000000000| delays: -
  m0, b02: |11111000000000000000000000000000| delays: 02+-02
  m0, b03: |00000011111111111111000000000000| delays: 12+-07
  m0, b04: |00000000000000000000001111111111| delays: 27+-04
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
[... 76 lines cut ...]
sL5DdSMmkekro
Timeout
Booting from SDCard in SD-Mode...
Booting from boot.json...
Booting from boot.bin...
SDCard boot failed.
Ethernet phy mode: GMII
Booting from network...
Local IP: 192.168.1.50
Remote IP: 192.168.1.100
Booting from boot.json...
Booting from boot.bin...
Copying boot.bin to 0x40000000... 
Network boot failed.
No boot medium found

--============= Console ================--
litex> 
```

Here is the bistream in case you want to test on your board:
[aliexpress_stlv7325_2022_04_27.zip](https://github.com/enjoy-digital/litedram/files/8572109/aliexpress_stlv7325_2022_04_27.zip)

And a picture of the SODIMM memory mounted on the board:
[image]

developandplay, 2022-04-27:
I'm using a fresh OpenOCD build so that flashing with OpenOCD works.
It seems like the `stream.cfg` that Litex generates is not up to date with that OpenOCD-0.11.0.
As a workaround I set debug mode to `-d0` to disable warnings here: https://github.com/enjoy-digital/litex/blob/665367fe67d8985d3d0e327c790bfa9dee5e0e0a/litex/build/openocd.py#L189

```
Warn : DEPRECATED! (stream.cfg:20) use 'expr { 0x0061 & 0x200 }' not 'expr 0x0061 & 0x200'
Warn : DEPRECATED! (stream.cfg:21) use 'expr { 0x0061 & 0 }' not 'expr 0x0061 & 0'
Warn : DEPRECATED! (stream.cfg:31) use 'expr { 0 && (0 > 0) && (0 < 4096) }' not 'expr 0 && (0 > 0) && (0 < 4096)'
```

Unfortunately even with your bitstream the initialization fails.
```
Memtest at 0x40000000 (2.0MiB)...
  Write: 0x40000000-0x40200000 2.0MiB     
   Read: 0x40000000-0x40200000 2.0MiB     
  bus errors:  75/256
  addr errors: 0/8192
  data errors: 523151/524288
Memtest KO
Memory initialization failed
```

I'll order some new memory and try it with that.

developandplay, 2022-04-27:
@enjoy-digital 
From what I can tell it just seems like we have different ddr3 memory modules.
How do I go about getting from my part number to the base chip and its timings?

developandplay, 2022-04-28:
I entered the information from the reference manual but it seems like that doesn't make any difference.
So I just ordered the same memory module as you are using. Let's see if that makes a difference or my board is faulty.

enjoy-digital, 2022-04-28:
It will be easier to provide support yes :)

pftbest, 2022-09-30:
I have the same Samsung SO-DIMM module as @enjoy-digital and for me memtest from litex_bios passes successfully. But I have an issue when using faster CPU cores which take advantage of memory_bus for direct connection to litedram.

I can reliably reproduce results in this table, I tried each configuration more than once
| CPU | mem_test (litex) | memtester (GPLv2) |
| - | -| -|
| VexRiscV | OK | OK |
| VexRiscV-SMP | OK | OK |
| Rocket-linux | OK | FAIL |
| NaxRiscV | OK | FAIL |

https://twitter.com/pftbest/status/1575612402153816064

I merged the memtester code in the litex_bios so you don't need to run full linux to try it
https://github.com/pftbest/litex/commit/c869a04805921a5a9061b8c8ab951f4e8583292d (GPLv2)

pftbest, 2022-09-30:
Full logs:
[VexRiscvSMP-OK.txt](https://github.com/enjoy-digital/litedram/files/9687424/VexRiscvSMP-OK.txt)
[Rocket-1-FAIL.txt](https://github.com/enjoy-digital/litedram/files/9687425/Rocket-1-FAIL.txt)
[Rocket-2-FAIL.txt](https://github.com/enjoy-digital/litedram/files/9687426/Rocket-2-FAIL.txt)
[NaxRiscv-FAIL.txt](https://github.com/enjoy-digital/litedram/files/9687601/NaxRiscv-FAIL.txt)

Dolu1990, 2022-09-30:
and i guess NaxRiscv with xlen=32 pass ?

Dolu1990, 2022-10-01:
(you had xlen=64 right ?)

pftbest, 2022-10-01:
@Dolu1990  Good idea! I didn't think I can change it to 32bits.

Yes, you are right NaxRiscv with xlen=32 works OK, no errors at all.
So it seems memory corruption only happens on 64bit CPUs, very strange. I'll update the table.

pftbest, 2022-10-01:
@Dolu1990 @enjoy-digital I found a fix for this issue.

I compared the XDC file with the one from reference design and found some differences. When I changed platform to match the reference design the problem was fixed.

I tested the change with Rocket-linuxd and 64bit NaxRiscv and both work fine. I will run longer tests today just to be 100% sure.

PR: https://github.com/litex-hub/litex-boards/pull/430

Dolu1990, 2022-10-01:
Nice :D

gillham, 2022-10-01:
Sweet, thanks for finding those!  I saw that AC12 when first working on the board in some other GitHub project but the seller’s documentation didn’t list it for this board.  I’m glad you knew more about this than I did.

pftbest, 2022-10-01:
@gillham I got the full schematics from the seller. If you like I can send it to your email.

AC12 is address bit 15, it's not used for 1GB stick. But it's still wired to the SO-DIMM slot, so I add it just in case.

pftbest, 2022-10-01:
schematic: https://drive.google.com/file/d/13DcdxPHt6-HoCxs3twyx_PKvQvdvym97/view?usp=sharing

yyx32, 2023-04-23:
Hello everyone, I am using a self-made FPGA board, and the FPGA chip is also using 325T (specifically xc7k325tffg900-2). I am using DDR3 with  DDR3_ SDRAM/SODIMMs/MT8KTF51264HZ-1G9（4GB）, based on several existing examples, my XML file is as follows:
```
{
    # General ------------------------------------------------------------------
    "speedgrade": -2,          # FPGA speedgrade
    "cpu":        "vexriscv",  # CPU type (ex vexriscv, serv, None)
    "memtype":    "DDR3",      # DRAM type
    "uart":       "rs232",     # Type of UART interface (rs232, fifo)

    # PHY ----------------------------------------------------------------------
    "cmd_latency":     1,             # Command additional latency
    "sdram_module":    "MT8KTF51264", # SDRAM modules of the board or SO-DIMM
    "sdram_module_nb": 8,             # Number of byte groups
    "sdram_rank_nb":   1,             # Number of ranks
    "sdram_phy":       "K7DDRPHY",    # Type of FPGA PHY

    # Electrical ---------------------------------------------------------------
    "rtt_nom": "60ohm",  # Nominal termination
    "rtt_wr":  "60ohm",  # Write termination
    "ron":     "34ohm",  # Output driver impedance

    # Frequency ----------------------------------------------------------------
    "input_clk_freq":   100e6, # Input clock frequency
    "sys_clk_freq":     100e6, # System clock frequency (DDR_clk = 4 x sys_clk)
    "iodelay_clk_freq": 200e6, # IODELAYs reference clock frequency

    # Core ---------------------------------------------------------------------
    "cmd_buffer_depth": 16,    # Depth of the command buffer

    # User Ports ---------------------------------------------------------------
    "user_ports": {
        # "axi_0" : {
        #     "type": "axi",
        #     "id_width": 4,
        #     "data_width": 64,
        # },
        # "wishbone_0" : {
        #     "type":  "wishbone",
        #     "block_until_ready": True,
        # },
        # "native_0" : {
        #     "type": "native",
        # },
        # "fifo_0" : {
        #     "type":  "fifo",
        #     "base":  0x00000000,
        #     "depth": 0x01000000,
        # },
    },
}
```
At the same time, I also constrained INTERNATIONAL_ VREF, as follows:
`set_property INTERNAL_VREF  0.75 [get_iobanks 33]`
But my DDR3 test still failed with an error:
```
--============= [1mConsole[0m ================--

[92;1mlitex[0m> 
[1m        __   _ __      _  __[0m
[1m       / /  (_) /____ | |/_/[0m
[1m      / /__/ / __/ -_)>  <[0m
[1m     /____/_/\__/\__/_/|_|[0m
[1m   Build your hardware, easily![0m

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 23 2023 11:30:04
 BIOS CRC passed (67674a4c)

[... 140 lines cut ...]
   Read: 0x40000000-0x40080000 512.0KiB   
   Read: 0x40000000-0x400a0000 640.0KiB   
   Read: 0x40000000-0x400c0000 768.0KiB   
   Read: 0x40000000-0x400e0000 896.0KiB   
   Read: 0x40000000-0x40100000 1.0MiB   
   Read: 0x40000000-0x40120000 1.1MiB   
   Read: 0x40000000-0x40140000 1.2MiB   
   Read: 0x40000000-0x40160000 1.3MiB   
   Read: 0x40000000-0x40180000 1.5MiB   
   Read: 0x40000000-0x401a0000 1.6MiB   
   Read: 0x40000000-0x401c0000 1.7MiB   
   Read: 0x40000000-0x401e0000 1.8MiB   
   Read: 0x40000000-0x40200000 2.0MiB   
  bus errors:  32/256
  addr errors: 0/8192
  data errors: 261081/524288
Memtest KO
Memory initialization failed

--============= [1mConsole[0m ================--

[92;1mlitex[0m> 
```
Does anyone know what may be causing this, or has anyone ever used the same model of DDR3
@enjoy-digital

enjoy-digital, 2023-04-23:
Hi @yyx32,

the issue seems related to module m1 (upper byte of your first  16-bit memory module). This could be useful to enable addtionnal debug in `liblitedram` and look at m1 logs. Are you also sure the DRAM is only one FPGA bank for the DRAM? (64-bit generally uses at least two).

yyx32, 2023-04-23:
Thank you, as I am not familiar with DDR yet, I will check what you said and provide feedback. Thank you again @enjoy-digital

yyx32, 2023-04-23:
In addition， what you mean is the value of sdram_rank_nb parameter at least 2 (according to some data, the number of banks is generally 8) @enjoy-digital

yyx32, 2023-04-24:
Finally, this is the MIG IP XML file corresponding to this DDR:
```
<?xml version="1.0" encoding="UTF-8" standalone="no" ?>
<Project NoOfControllers="1">

  <!-- IMPORTANT: This is an internal file that has been generated by the MIG software. Any direct editing or changes made to this file may result in unpredictable behavior or data corruption. It is strongly advised that users do not edit the contents of this file. Re-run the MIG GUI with the required settings if any of the options provided below need to be altered. -->

  <ModuleName>mig_7series_0</ModuleName>

  <dci_inouts_inputs>1</dci_inouts_inputs>

  <dci_inputs>1</dci_inputs>

  <Debug_En>OFF</Debug_En>

  <DataDepth_En>1024</DataDepth_En>

  <LowPower_En>ON</LowPower_En>

  <XADC_En>Enabled</XADC_En>

  <TargetFPGA>xc7k325t-ffg900/-2</TargetFPGA>

  <Version>4.2</Version>

  <SystemClock>Differential</SystemClock>

  <ReferenceClock>Use System Clock</ReferenceClock>

  <SysResetPolarity>ACTIVE LOW</SysResetPolarity>

  <BankSelectionFlag>FALSE</BankSelectionFlag>

  <InternalVref>0</InternalVref>

  <dci_hr_inouts_inputs>40 Ohms</dci_hr_inouts_inputs>

  <dci_cascade>1</dci_cascade>

  <Controller number="0">
    <MemoryDevice>DDR3_SDRAM/SODIMMs/MT8KTF51264HZ-1G9</MemoryDevice>
    <TimePeriod>2500</TimePeriod>
    <VccAuxIO>1.8V</VccAuxIO>
    <PHYRatio>2:1</PHYRatio>
    <InputClkFreq>200</InputClkFreq>
    <UIExtraClocks>0</UIExtraClocks>
    <MMCM_VCO>800</MMCM_VCO>
    <MMCMClkOut0> 1.000</MMCMClkOut0>
    <MMCMClkOut1>1</MMCMClkOut1>
    <MMCMClkOut2>1</MMCMClkOut2>
    <MMCMClkOut3>1</MMCMClkOut3>
    <MMCMClkOut4>1</MMCMClkOut4>
    <DataWidth>64</DataWidth>
    <DeepMemory>1</DeepMemory>
    <DataMask>1</DataMask>
    <ECC>Disabled</ECC>
    <Ordering>Normal</Ordering>
    <BankMachineCnt>4</BankMachineCnt>
    <CustomPart>FALSE</CustomPart>
    <NewPartName></NewPartName>
    <RowAddress>16</RowAddress>
    <ColAddress>10</ColAddress>
    <BankAddress>3</BankAddress>
    <MemoryVoltage>1.5V</MemoryVoltage>
    <UserMemoryAddressMap>BANK_ROW_COLUMN</UserMemoryAddressMap>
    <PinSelection>
      <Pin IN_TERM="" IOSTANDARD="SSTL15" PADName="AB9" SLEW="FAST" VCCAUX_IO="NORMAL" name="ddr3_addr[0]"/>
      <Pin IN_TERM="" IOSTANDARD="SSTL15" PADName="AA11" SLEW="FAST" VCCAUX_IO="NORMAL" name="ddr3_addr[10]"/>
      <Pin IN_TERM="" IOSTANDARD="SSTL15" PADName="Y11" SLEW="FAST" VCCAUX_IO="NORMAL" name="ddr3_addr[11]"/>
      <Pin IN_TERM="" IOSTANDARD="SSTL15" PADName="AB8" SLEW="FAST" VCCAUX_IO="NORMAL" name="ddr3_addr[12]"/>
[... 125 lines cut ...]
    <mrBurstType name="Read Burst Type and Length">Sequential</mrBurstType>
    <mrCasLatency name="CAS Latency">6</mrCasLatency>
    <mrMode name="Mode">Normal</mrMode>
    <mrDllReset name="DLL Reset">No</mrDllReset>
    <mrPdMode name="DLL control for precharge PD">Slow Exit</mrPdMode>
    <emrDllEnable name="DLL Enable">Enable</emrDllEnable>
    <emrOutputDriveStrength name="Output Driver Impedance Control">RZQ/7</emrOutputDriveStrength>
    <emrMirrorSelection name="Address Mirroring">Disable</emrMirrorSelection>
    <emrCSSelection name="Controller Chip Select Pin">Enable</emrCSSelection>
    <emrRTT name="RTT (nominal) - On Die Termination (ODT)">RZQ/4</emrRTT>
    <emrPosted name="Additive Latency (AL)">0</emrPosted>
    <emrOCD name="Write Leveling Enable">Disabled</emrOCD>
    <emrDQS name="TDQS enable">Enabled</emrDQS>
    <emrRDQS name="Qoff">Output Buffer Enabled</emrRDQS>
    <mr2PartialArraySelfRefresh name="Partial-Array Self Refresh">Full Array</mr2PartialArraySelfRefresh>
    <mr2CasWriteLatency name="CAS write latency">5</mr2CasWriteLatency>
    <mr2AutoSelfRefresh name="Auto Self Refresh">Enabled</mr2AutoSelfRefresh>
    <mr2SelfRefreshTempRange name="High Temparature Self Refresh Rate">Normal</mr2SelfRefreshTempRange>
    <mr2RTTWR name="RTT_WR - Dynamic On Die Termination (ODT)">Dynamic ODT off</mr2RTTWR>
    <PortInterface>NATIVE</PortInterface>
  </Controller>

</Project>
```
If possible, could you help me point out the errors in the yml file? thank you

enjoy-digital, 2023-04-24:
Hi @yyx32,

what I mean is that you are only constraining `INTERNAL_VREF` on Bank 33, while when looking at https://www.xilinx.com/content/dam/xilinx/support/packagefiles/k7packages/xc7k325tffg900pkg.txt and the `.xml `you provided, the DRAM pin are on Bank 32, 33, 34, so the 3 banks should have the `INTERNAL_VREF` constraints. Can you test this? With the log you provided, there are good chance this is the issue.

yyx32, 2023-04-25:
Hi @enjoy-digital 
Additionally, I attempted to convert  The memory type of the sitlinv_stlv7325 development board is changed to MT8K51264, and the code is as follows:
```
# DDR3 SDRAM -------------------------------------------------------------------------------
    if not self.integrated_main_ram_size:
        self.ddrphy = s7ddrphy.K7DDRPHY(platform.request("ddram"),
            memtype      = "DDR3",
            nphases      = 4,
            sys_clk_freq = sys_clk_freq,
        )
        self.add_sdram("sdram",
            phy           = self.ddrphy,
            module        = MT8KTF51264(sys_clk_freq, "1:4"),
            l2_cache_size = kwargs.get("l2_size", 8192),
        )

```
When I run the command:
```
../litex-boards/litex_boards/targets/sitlinv_stlv7325.py --build --load --cpu-type vexriscv --sys-clk-freq 50e6 --with-sdcard --with-ethernet --csr-csv=csr.csv
```
The error is as follows:
```
INFO:SoC:RAM sram added Origin: 0x10000000, Size: 0x00002000, Mode: RWX, Cached: True Linker: False.
INFO:SoCIRQHandler:uart IRQ allocated at Location 0.
INFO:SoCIRQHandler:timer0 IRQ allocated at Location 1.
INFO:SoCBusHandler:main_ram Region added at Origin: 0x40000000, Size: 0x100000000, Mode: RWX, Cached: True Linker: False.
INFO:SoCBusHandler:main_ram added as Bus Slave.
INFO:SoCBusHandler:Allocating IO Region of size 0x00002000...
ERROR:SoCBusHandler:Not enough Address Space to allocate Region.
```
How can I fix it?Thank you

enjoy-digital, 2023-04-25:
Hi @yyx32,

you'll probably have to limit size to 1GB as done for example here: https://github.com/litex-hub/litex-boards/blob/master/litex_boards/targets/sqrl_xcu1525.py#L84

enjoy-digital, 2023-04-26:
The limitation will apply to LiteX, that will only map 1GB instead of the 4GB. So you can just apply the change and test. The 1GB limitation will be removed at some point, but this still need to be done.

yyx32, 2023-04-26:
ok,thank you

yyx32, 2023-04-27:
Hello, I have replaced the model of DDR3 with the same 1G DDR3, as follows:
[image]
At the same time, I successfully passed the DDR3 read and write test on the stlv7325 development board, but when I tested using this DDR3 on my own development board (I tried to use the same DDR configuration as much as possible), I still encountered error as follows：
```
--============= [1mConsole[0m ================--

[92;1mlitex[0m> 
[1m        __   _ __      _  __[0m
[1m       / /  (_) /____ | |/_/[0m
[1m      / /__/ / __/ -_)>  <[0m
[1m     /____/_/\__/\__/_/|_|[0m
[1m   Build your hardware, easily![0m

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on Apr 27 2023 16:00:42
 BIOS CRC passed (ac318310)

 LiteX git sha1: b367c271

--=============== [1mSoC[0m ==================--
[1mCPU[0m:		VexRiscv @ 100MHz
[1mBUS[0m:		WISHBONE 32-bit @ 4GiB
[1mCSR[0m:		32-bit data
[1mROM[0m:		48.0KiB
[1mSRAM[0m:		8.0KiB
[1mL2[0m:		128B
[1mSDRAM[0m:		1.0GiB 64-bit @ 800MT/s (CL-6 CWL-5)
[1mMAIN-RAM[0m:	16.0MiB

--========== [1mInitialization[0m ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 32
  Cmd/Clk scan (0-16)
  |0001110000000001| best: 15
  Setting Cmd/Clk delay to 15 taps.
  Data scan:
  m0: |111111111100000000000000| delay: 00
  m1: |111111111000000000000000| delay: 00
  m2: |111111111111000000000000| delay: 00
  m3: |111111111111100000000000| delay: 00
  m4: |000001111111111111111000| delay: 05
  m5: |000000111111111111111000| delay: 06
  m6: |000000011111111111111100| delay: 07
  m7: |000000111111111111111100| delay: 06
Write latency calibration:
m0:0 m1:0 m2:0 m3:0 m4:0 m5:0 m6:0 m7:0 
Write DQ-DQS training:
m0: |111111110000000000000000000000000| delays: 03+-03
m1: |000000000000000000000000000000000| delays: -
m2: |111111100000000000000000000000000| delays: 02+-02
m3: |111111100000000000000000000000000| delays: 02+-02
m4: |111111111111000000000000000000000| delays: 05+-05
m5: |111111111111000000000000000000000| delays: 05+-05
m6: |111111111111111000000000000000000| delays: 06+-06
m7: |111111111111110000000000000000000| delays: 06+-06
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00001111111111111000000000000000| delays: 10+-06
  m0, b02: |00000000000000000000111111111111| delays: 25+-05
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 10+-06
  m1, b00: |00000000000000000000000000000000| delays: -
[... 86 lines cut ...]
   Read: 0x40000000-0x40060000 384.0KiB   
   Read: 0x40000000-0x40080000 512.0KiB   
   Read: 0x40000000-0x400a0000 640.0KiB   
   Read: 0x40000000-0x400c0000 768.0KiB   
   Read: 0x40000000-0x400e0000 896.0KiB   
   Read: 0x40000000-0x40100000 1.0MiB   
   Read: 0x40000000-0x40120000 1.1MiB   
   Read: 0x40000000-0x40140000 1.2MiB   
   Read: 0x40000000-0x40160000 1.3MiB   
   Read: 0x40000000-0x40180000 1.5MiB   
   Read: 0x40000000-0x401a0000 1.6MiB   
   Read: 0x40000000-0x401c0000 1.7MiB   
   Read: 0x40000000-0x401e0000 1.8MiB   
   Read: 0x40000000-0x40200000 2.0MiB   
  bus errors:  32/256
  addr errors: 0/8192
  data errors: 261087/524288
Memtest KO
Memory initialization failed

--============= [1mConsole[0m ================--

[92;1mlitex[0m> 
```
It's very important for me to help me see where the mistake was again. Thank you @enjoy-digital

yyx32, 2023-05-05:
Now, I have encountered a new problem. After testing DDR3 on the stlv7325 development board (without AXI interface), I opened the AXI interface in the controller and connected it to my application project. The DDR part of the project is as follows:
[image]
Except for adding the AXI interface, other configurations have not changed, but the DDR initialization test failed and the error is as follows:
```
[1m        __   _ __      _  __[0m
[1m       / /  (_) /____ | |/_/[0m
[1m      / /__/ / __/ -_)>  <[0m
[1m     /____/_/\__/\__/_/|_|[0m
[1m   Build your hardware, easily![0m

 (c) Copyright 2012-2023 Enjoy-Digital
 (c) Copyright 2007-2015 M-Labs

 BIOS built on May  4 2023 15:16:49
 BIOS CRC passed (1d24e80b)

 LiteX git sha1: b367c271

--=============== [1mSoC[0m ==================--
[1mCPU[0m:		VexRiscv @ 100MHz
[1mBUS[0m:		WISHBONE 32-bit @ 4GiB
[1mCSR[0m:		32-bit data
[1mROM[0m:		48.0KiB
[1mSRAM[0m:		8.0KiB
[1mL2[0m:		128B
[1mSDRAM[0m:		1.0GiB 64-bit @ 800MT/s (CL-6 CWL-5)
[1mMAIN-RAM[0m:	16.0MiB

--========== [1mInitialization[0m ============--
Initializing SDRAM @0x40000000...
Switching SDRAM to software control.
Write leveling:
  tCK equivalent taps: 32
  Cmd/Clk scan (0-16)
  |0000000000000001| best: 6
  Setting Cmd/Clk delay to 6 taps.
  Data scan:
  m0: |000000000000000000111111| delay: 18
  m1: |100000000000000000111111| delay: 18
  m2: |111100000000000000000111| delay: -
  m3: |111100000000000000000011| delay: -
  m4: |111111111100000000000000| delay: 00

  m5: |111111111000000000000000| delay: 00
  m6: |111111111111000000000000| delay: 00
  m7: |111111111111000000000000| delay: 00
Write latency calibration:
m0:2 m1:2 m2:0 m3:0 m4:0 m5:0 m6:0 m7:0 
Write DQ-DQS training:
m0: |000000000011111111111111100000000| delays: 16+-06
m1: |000000000001111111111111110000000| delays: 17+-06
m2: |111111110000000000000000000000000| delays: 03+-03
m3: |111111100000000000000000000000000| delays: 02+-02
m4: |111111110000000000000000000000000| delays: 03+-03
m5: |111111100000000000000000000000000| delays: 02+-02
m6: |111111110000000000000000000000000| delays: 03+-03
m7: |111111110000000000000000000000000| delays: 03+-03
Read leveling:
  m0, b00: |00000000000000000000000000000000| delays: -
  m0, b01: |00000000000000000111111111110000| delays: 22+-05
  m0, b02: |00000000000000000000000000000000| delays: -
  m0, b03: |00000000000000000000000000000000| delays: -
  m0, b04: |00000000000000000000000000000000| delays: -
  m0, b05: |00000000000000000000000000000000| delays: -
  m0, b06: |00000000000000000000000000000000| delays: -
  m0, b07: |00000000000000000000000000000000| delays: -
  best: m0, b01 delays: 23+-05
  m1, b00: |00000000000000000000000000000000| delays: -
  m1, b01: |00000000000000000011111111111100| delays: 23+-05
  m1, b02: |00000000000000000000000000000000| delays: -
[... 82 lines cut ...]
   Read: 0x40000000-0x40060000 384.0KiB   
   Read: 0x40000000-0x40080000 512.0KiB   
   Read: 0x40000000-0x400a0000 640.0KiB   
   Read: 0x40000000-0x400c0000 768.0KiB   
   Read: 0x40000000-0x400e0000 896.0KiB   
   Read: 0x40000000-0x40100000 1.0MiB   
   Read: 0x40000000-0x40120000 1.1MiB   
   Read: 0x40000000-0x40140000 1.2MiB   
   Read: 0x40000000-0x40160000 1.3MiB   
   Read: 0x40000000-0x40180000 1.5MiB   
   Read: 0x40000000-0x401a0000 1.6MiB   
   Read: 0x40000000-0x401c0000 1.7MiB   
   Read: 0x40000000-0x401e0000 1.8MiB   
   Read: 0x40000000-0x40200000 2.0MiB   
  bus errors:  128/256
  addr errors: 0/8192
  data errors: 262144/524288
Memtest KO
Memory initialization failed

--============= [1mConsole[0m ================--

[92;1mlitex[0m> 
```
It should be noted that the project is still ran on the stlv7325 development board, and it runs normally when using MIG as the DDR3 controller. What could be the reason for the failure of the DDR3 controller initialization after adding the AXI interface? @enjoy-digital

yyx32, 2023-05-06:
Finally, it was found that after multiple resets after accessing the application project, it may be successfully initialized once in five or six times, indicating that the work is very unstable

yyx32, 2023-05-09:
I have increased the stability of the controller by using Vivado's pblock function, but I have found that even if DDR initialization is successful, the Linux system cannot be started. Do I need to adapt the driver of the DDR controller？ @enjoy-digital