enjoy-digital/litedram #293: Initialization failed on Artix after e5e3b6c

kaolpr, 2022-02-22:
We've updated LiteX and dependencies lately and it turned out that our Artix-based board started failing on memory (SDRAM) initialization. After some cross checking and building with different commits of litedram we've found that that commit https://github.com/enjoy-digital/litedram/commit/e5e3b6c9a3f9108e2db63ce21f111fe131199878 was the problem.

It seems that `write_latency_calibration` should always be `True` for S7 DDR Phy as physical elements that `sdram_write_latency_calibration` ([sdram.c](https://github.com/enjoy-digital/litex/blob/4acbafbc19fe5ec91e7a02f7ae9494ba6fca5bdc/litex/soc/software/liblitedram/sdram.c#L1012)) function interacts with are present regardless of `with_odelay` parameter.

Setting `write_latency_calibration = True` in master version of litedram made initialization successful again.

@enjoy-digital Can you please confirm and, if my findings are right, revert changes introduced by https://github.com/enjoy-digital/litedram/commit/e5e3b6c9a3f9108e2db63ce21f111fe131199878 ?

enjoy-digital, 2022-02-23:
Hi @kaolpr,

thanks for the feedback, can you tell me a bit more about the Artix-based board that are failing? Are they off-the-shelf boards or custom ones?

I did https://github.com/enjoy-digital/litedram/commit/e5e3b6c9a3f9108e2db63ce21f111fe131199878 doing the  write latency calibration was making calibration fail on some boards (ex Acorn CLE215+). We could also do it a bit differently and allow user to configure this (And enabled by default). This would  solves your issue and would allow me to also disable it on the problematic board. But if you are seeing this behavior on an off-the-shelf board (that I maybe also have), I could investigate more.

kaolpr, 2022-02-23:
I was using [AFC v4.0](https://ohwr.org/project/afc/wikis/home). There are ~10 units of this revision running in different institutions and we have no information on hardware issues related to SDRAM (though I believe we're using it with LiteX only at Technosystem at the moment).

enjoy-digital, 2022-02-23:
Thanks @kaolpr. I just exposed the `write_latency_calibration` parameter and set it to True by default. This will allow me to disable it on Artix7 boards where it is problematic.

enjoy-digital, 2022-02-23:
Exposed and reverted to True by default with https://github.com/enjoy-digital/litedram/commit/e48471ea43c82a473d4b58bda32760b852d776a7.

closed 2022-02-23 by enjoy-digital