enjoy-digital/litedram #299: ULX4M DM signal is not connected to DQS group

goran-mahovlic, 2022-03-21:
Hi, 

I am trying to get DDR3 running on ULX4M, I am currently using gsd_butterstick.py with all changes needed for ULX4M, but I did not know that DM also needs to be in the same DQS group so I have changed this 

https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/ecp5ddrphy.py#L339

to use

ODDRX2F

But I get memtest fail, so maybe you know out of your hear should this be ok quick fix, and how could I do more debugging to get more info what is wrong if it is not because DM signals.

[image]

enjoy-digital, 2022-03-21:
Hi @goran-mahovlic,

the BIOS `mem_test `is only doing full access to the DRAM (no use of DM), so in a first time, can you comment this: https://github.com/enjoy-digital/litedram/blob/master/litedram/phy/ecp5ddrphy.py#L339-L346 and replace it with:
`self.comb += pads.dm[i].eq(0)`

This will avoid latency related issue on DM and will help knowing if the issue is related to DM or something else.

goran-mahovlic, 2022-03-21:
> Hi @goran-mahovlic,
>
> the BIOS `mem_test `is only doing full access to the DRAM (no use of DM), so in a first time, can you comment this: [...]

Thank you, that works!

[image]

closed 2022-03-21 by goran-mahovlic