enjoy-digital/litedram #304: Strategy steering: Best way to stream data into a custom accelerator when litedram is being used inside Litex SoC?

francis2tm, 2022-04-04:
Hello,
My goal is to have a litex SoC with a vexriscv CPU and a custom accelerator that constantly consumes memory from an external DDR3. What is the best way to achieve "stream" like dataflow onto the accelerator, considering that the CPU will also be using the DDR3 since all its memory regions reside in the DDR3 (text, data, bss, heap and stack)? Essentially the accelerator will have priority for everything since it's okay to sacrifice CPU's performance.
My first approach is to connect the accelerator to the wishbone bus via a DMA. However, this would mean that the CPU would race with DMA for the BUS pretty much all the time. Is there a way to get more "direct" connection to liteDRAM and thus improve accelerator's performance?

Thanks in advance

enjoy-digital, 2022-04-04:
This will provide better performance than connecting the DMAs to the Wishbone bus yes. LiteDRAM will automatically arbitrate the ports yes (with a Round Robin arbiter). It would be possible to add a priority control signal if required.

francis2tm, 2022-04-04:
thanks a lot for the info!

closed 2022-04-04 by francis2tm