enjoy-digital/litedram #306: Why are CL and CWL not included as speedgrade parameters in the module class?

jaccharrison, 2022-04-15:
Hello,

I've defined several module classes for different DDR4 DIMMs. I've noticed that not all speedgrade-relevant timing parameters are included in the `speedgrade_timings` field of the `DDR4Module` class. For example, tCCD and tRRD are currently defined in `technology_timings`, but in examining DRAM datasheets I've found that these parameters change depending on speed grade. More importantly, CL and CWL, which are timing-specific parameters and must be passed to the PHY, are not associated with the module class. I think it would make sense for these to be included in a `DDR4Module` because if they are not, these parameters must be manually managed for each instantiation of the memory controller.

Is this intentional? Would a pull request changing this behavior be appreciated?

Cheers,
Jacob

enjoy-digital, 2022-04-20:
Thanks @jaccharrison. I'll have a closer look, the reason is probably just related to the history of the development. Your fresh eyes probably just catched inconsistencies we no longer see :) 

Moving most of https://github.com/enjoy-digital/litedram/blob/master/litedram/common.py#L20-L78 to modules could indeed make sense.

enjoy-digital, 2026-11-06:
Sorry for the very late answer. This is still how it works: CL/CWL defaults come from `common.py` (the PHYs accept `cl`/`cwl` overrides), and tCCD/tRRD stay in `technology_timings`. There is no technical reason beyond history, but I can't plan this on the project side. A PR would be welcome. Closing as not planned.

closed 2026-11-06 by enjoy-digital