enjoy-digital/litedram #308: QuarterRateGENSDRPHY

machdyne, 2022-06-25:
I'm trying to implement a Quarter-rate Generic SDR PHY module so that I can run a CPU at 40MHz with the SDRAM at 160MHz. It looked straight-forward at first but I don't really know what I'm doing and haven't had much success. If anyone could provide any tips or examples of how to implement this it would be appreciated.

machdyne, 2022-09-05:
I mashed together the code from HalfRateGENSDRPHY and S6QuarterRateDDRPHY, and it's not working, the memtest fails, but maybe somebody can tell me where I'm going wrong?

https://github.com/machdyne/litedram/commit/f9da309ed41c9b0bf726c12a18f53d235c61a547

rowanG077, 2022-09-07:
Your best bet is to make a PR of your changes to this repo that makes it easier to look at the diff and will (hopefully) be merged.

enjoy-digital, 2026-11-06:
Sorry for the late answer. There is no quarter-rate PHY and it isn't planned on the project side, but `HalfRateGENSDRPHY` was improved this week, including a CL3 fix, so 80 MHz sys with 160 MHz SDRAM is the closest option. On your commit, two things look wrong, though I haven't run it: `ClockDomainsRenamer("sys4x")` renames only `sys`, so the half-rate PHY's `sys2x` stays slower than its `sys`, and a 1-bit `phase_sel` can't select 4 phases. Happy to review a PR. Closing as not planned.

closed 2026-11-06 by enjoy-digital