enjoy-digital/litedram #309: Simulation issue, Arty S7

TheAnimatrix, 2022-10-04:
I've generated a DDR3 core with
`litedram_gen examples/arty.yml`

my arty.yml is as follows
```
{
    # General ------------------------------------------------------------------
    "speedgrade": -1,          # FPGA speedgrade
    "cpu":        "None",  # CPU type (ex vexriscv, serv, None)
    "memtype":    "DDR3",      # DRAM type

    # PHY ----------------------------------------------------------------------
    "cmd_latency":     0,             # Command additional latency
    "sdram_module":    "MT41K64M16", # SDRAM modules of the board or SO-DIMM
    "sdram_module_nb": 2,             # Number of byte groups
    "sdram_rank_nb":   1,             # Number of ranks
    "sdram_phy":       "A7DDRPHY",    # Type of FPGA PHY

    # Electrical ---------------------------------------------------------------
    "rtt_nom": "60ohm",  # Nominal termination
    "rtt_wr":  "60ohm",  # Write termination
    "ron":     "34ohm",  # Output driver impedance

    # Frequency ----------------------------------------------------------------
    "input_clk_freq":   100e6, # Input clock frequency
    "sys_clk_freq":     100e6, # System clock frequency (DDR_clk = 4 x sys_clk)
    "iodelay_clk_freq": 200e6, # IODELAYs reference clock frequency

    # Core ---------------------------------------------------------------------
    "cmd_buffer_depth": 16,    # Depth of the command buffer

    # User Ports ---------------------------------------------------------------
    "user_ports": {
        "native_0" : {
            "type": "native",
            "block_until_ready": False,
        }
    },
}
```

I find that when i simulate the output module litedram_core.v with vivado and a simple testbench which just deasserts reset after 1000ns, that init_done and init_error are never asserted.

I've simulated 600us and all the module does is Precharge and Refresh every 7-8us. Where is the mode register initialization ? ZQ Calibration etc ? Do i need to generate a cpu for this ? How can i do it without a cpu ?

Am i missing something ? I'm very new here so i'm still trying to figure stuff out, help is appreciated. Thanks.

TheAnimatrix, 2022-10-04:
I've simulated it with a ddr3 model i got from micron.

rowanG077, 2022-10-04:
The standalone core needs you to manually run the initialization sequence.

TheAnimatrix, 2022-10-04:
How do i do that ?

rowanG077, 2022-10-04:
The initialization is usually done by software running on a softcore. The code that does it is here: https://github.com/enjoy-digital/litex/blob/master/litex/soc/software/liblitedram/sdram.c

TheAnimatrix, 2022-10-04:
If i were to initialize with the vexriscV cpu, how would i simulate the same ? Would it be automatic provided i load the litedram_core_rom.init into the cpu ?

I'm assuming litedram_core_rom.init is sdram.c but compiled

rowanG077, 2022-10-04:
Unfortunately I have never simulated the core so I can't really answer how to do it. If I were you I would try to understand the unit tests: https://github.com/enjoy-digital/litedram/tree/master/test 

They do simulate the core so it's definitely possible.

TheAnimatrix, 2022-10-05:
How do i generate a core with the smallest processor (serv?) just for calibrating the dram, i don't want anything else like the Ethernet PHY , SD card or anything. Should i change litex-boards/targets/arty-s7.py ? Also generating from the litex-boards section doesn't seem to generate the processor ? I only see a single verilog file with the dram controller.

TheAnimatrix, 2022-10-07:
For those who have similar beginner issues, feel free to contact me -> frustratymous@gmail.com

dinaabdelbaky, 2023-02-28:
@TheAnimatrix Could you please share your findings, I am facing the same issue and I am stuck, it would be great if you share how you managed to carry out the init_seq manually. Thanks in advance!