enjoy-digital/litedram #316: submodules verilog

CarrolXC, 2022-12-26:
Hi , I am generating verilog file through gen.py and I did see a top level verilog file in the "build" folder. However , I found I missed all the submodules used in the top level, like "BUFG"or "IOBUF" which I could trace to some other python file.Does anyone have ideas on how to generate all the verilog files needed including all the submodules.Sincerely appreciate any suggestion!

smosanu, 2023-01-02:
Some submodules like "BUFG" or "IOBUF" are specific to the tools for the board you use (in this case I assume it is the Xilinx tools?) and you shouldn't need to Verilog code for them...

enjoy-digital, 2023-01-10:
Hi @CarrolXC,

these primitives are indeed Xilinx's primitives with models available in Unisim: https://github.com/Xilinx/XilinxUnisimLibrary. You'll be able to simulate them with Xilinx simulator of by also compiling these models with your simulator (should work with iverilog, would need to be tested with Verilator).

enjoy-digital, 2026-11-06:
Closing as answered above: these are vendor primitives, with models in the Xilinx Unisim library. If you only need simulation, `litedram_gen --sim` gives a core without them.

closed 2026-11-06 by enjoy-digital